Ultimaker/Cura #11474: Reset button to get original Start-/Endgcode back in Printer settings

NilsRo, 2022-02-13:
### Is your feature request related to a problem?

More or less...I upgrading the Mega S printer profile continuously and changed the Start-/Endgcode also. But if somebody simply upgrading to a new Cura Version the Start-/Endcode is not changed. E.g. I fixed a bug and the endusers does not know it and run the old (or custom) Gcode forever.

### Describe the solution you'd like

Only some ideas:

- provide a reset button to get the original settings back
- provide a simple compare function like WinMerge (or very simple two text editors side by side :-) ) that users can compare changes 

### Describe alternatives you've considered

- Provide WinMerge with the installation (do not know if this is possible with the licence) but it is also Opensource and open the two versions there. Acutally I done it with creating a new printer and compare the initial Start-/Endgcode but I think many beginners would not know that.

### Affected users and/or printers

All printers, all users

GregValiant, 2022-02-13:
This raises an interesting point...and a philosophical question.

Let me just say that **_I hate_** when I screw-up when opening a 3mf file and my custom Start and End gcodes get over-written.  It would also bother me if Cura changed/updated them when I install a new version of Cura.  There are reasons I have altered them the way I have.  There are no bugs in my Startup or Ending gcode.

If there was a bug in the Start or End gcode of a printer definition file (and there has been at least one), and the bug was fixed in the next version of Cura - how would those "less savvy" users know?  
Should there be some mechanism in place to tell them _"The printer definition file for your "BrandX Doodle 3 Pro" has been altered.  Do you want to install the new definition?"_.  Would those people understand it or would it just be quickly dismissed?
My guess would be that a very large percentage of the casual Cura users have no idea that there is such a thing as "StartUp G-Code" or "End G-Code".  For a user such as myself (I would not call myself a "typical" end user) I have mine the way I want them and so I would always select "No".

If something like this were to be considered, maybe the option to "Install" another printer instance that would have the new definition could be an option.  Then a user could compare the newly installed BrandX Doodle #2 with their previously installed instance.

Then there is the code required to implement this.
Iterating through the 300+ printer definition files (most of which are 3rd party) to determine which ones have changed since the last version of Cura, and then going through the list of the user's installed printers to see if there is a match, I think would slow down the Cura installation a lot.  It already is no speed demon.

The philosophical point is - would Ultimaker (and we are talking "Ultimaker Cura") want to put the resources ($$$) into adding such a feature?  Where would be the benefit to Ultimaker?  Shouldn't it be on the third party manufacturer to inform their customers that something changed and that the customer needs to install a new printer instance in Cura? 

I can see certain benefits in this to a small group of end users.  I just don't see it happening.

NilsRo, 2022-02-13:
I think the new users will benefit as they simply want to use a printer. Perhaps a new idea: a checkbox with 'use printer default Start-/Endcode'

fvrmr, 2022-02-14:
Hi @NilsRo thank you for your request. 
I will bring it up with the team. Keep you posted.

fvrmr, 2022-02-18:
We have discussed it with the team and decided to create a ticket for this.
Devs see CURA-8974