Ultimaker/Cura #11476: Menu bar button has no functionality and cannot be disabled on macOS

KizzyCode, 2022-02-14:
### Application Version

4.13.1

### Platform

macOS Monterey 12.2.1 (21D62)

### Printer

Ender 3 Pro

### Reproduction steps

1. Minimize Cura

### Actual results

If I minimize Cura, a big bright blue menu bar button appears (see screenshot). However this icon has no functionality and cannot be disabled either.

[image]
 

### Expected results

The menu bar button should do something if I interact with it. Furthermore it should always be possible to disable the button entirely (e.g. via settings).

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

None

fvrmr, 2022-02-15:
Hi @KizzyCode thank you for your report.
I checked it on macOS Monetery 12.1 and there it also shows up and does nothing.
I will bring it up with the team. Keep you posted!

fvrmr, 2022-02-18:
I have discussed it with the team. And we decided to defer this issue, since it will not get a high priority for us to fix right now. 
I hope you understand.

KizzyCode, 2022-02-19:
Yeah of course; I mean this is just a minor UI inconvenience 😊

Ghostkeeper, 2022-02-26:
Duplicate of https://github.com/Ultimaker/Cura/issues/2687?

KizzyCode, 2022-02-26:
This could indeed be related, however what puzzles me is that this bug exists since 2017 whereas the macOS-variant happened just recently for the first time?

Ghostkeeper, 2022-03-14:
We didn't change anything here recently. But of course, MacOS also changes all the time.

closed 2022-09-08 by nallath