Ultimaker/Cura #11477: Paint on supports - Support blockers/Custom supports not enough

HermanGaviriaMarin, 2022-02-14:
### Is your feature request related to a problem?

Complex geometry with non uniform overhangs force users to place supports in very specific locations. Support blockers can be very handy but their implementation is very flawed when it comes to models where you need very precise support free locations, moreover, custom supports can be a huge paint to deal with, many times in the past their removal has been near impossible for me. Ultimately tree supports are the best support type but they cannot be routed to a specific location of the model without a ton of support blockers. 

### Describe the solution you'd like

PrusaSlicer has a feature called Paint on supports. This feature allows its users to designate an area of the model that will receive supports. This feature should be implemented into Ultimaker Cura, the only thing that would be different is that this feature would work with tree supports. 

I know what you will all say, "then why not go use PrusaSlicer if its such a key feature" to that I will say, cura is my preferred slicer, lots of great features and most importantly its what I'm used to using. Moreover, Prusaslicer will not have the arachne engine 2 which I've been using for the past months and its been great. 

### Describe alternatives you've considered

I've considered trying Prusa slicer but it would be pain to have to once again dial in all the setting that do work for me. I've also had to spend 20+ minutes working with support blockers and custom supports to get the correct support structures in my models. 

Paint on supports with tree supports would create an ease of work that would be unparalleled here in the ultimaker community.

### Affected users and/or printers

Ultimaker Cura Community.

closed 2022-02-14 by fvrmr

HermanGaviriaMarin, 2022-02-14:
Sorry for creating a duplicate. This seems to be a feature that has been requested since very long time ago. Is it even being worked on ? Will we ever see it come to fruition ?

fvrmr, 2022-02-14:
It is still on the backlog and we still would like to implement this, but there are more important things we want to implement at the moment. So it could take a while, since it is also a bit complex to implement as I understood correctly from the ticket we have on our backlog.