Ultimaker/Cura #11478: Some stl files when open complain about not being "Water tight"

peterj43, 2022-02-14:
### Application Version

4.13.1

### Platform

Ubuntu 21.04

### Printer

Enter 3 Pro

### Reproduction steps

1. Open certain stl files
2. Repeat opening the same file using  Cura 4.13.1 on windows 10 no issues are reported.
3. Opening the same file using PursaSlicer on Ubuntu 21.04  does not have an issue with the same file.

### Actual results

1. Mesh Tools box opens  " Model <filename.stl> is not watertight and may not print properly. Check X-ray view and repair.
2. No issues are reported.

### Expected results

I would expect the same result  on Ubuntu as Windows

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[Box insert base 2v4 v22.zip](https://github.com/Ultimaker/Cura/files/8061297/Box.insert.base.2v4.v22.zip)
This is the file which causes the problem

fieldOfView, 2022-02-14:
> Mesh Tools box opens " Model <filename.stl> is not watertight and may not print properly. "

That message is shown by the MeshTools plugin. Please report issues for the MeshTools plugin here instead:
https://github.com/fieldOfView/Cura-MeshTools/issues

fieldOfView, 2022-02-14:
(probably a duplicate of https://github.com/fieldOfView/Cura-MeshTools/issues/23, which will be fixed in the next release of the plugin)

closed 2022-02-15 by fvrmr