Ultimaker/Cura #11479: A setting that prevents the print head unnecessarily crossing open areas of the print.

johnnieq2, 2022-02-14:
### Is your feature request related to a problem?

OK, the object being printed is a single piece horseshoe shape printing vertically. (It's actually a fuselage for a RC model plane) The print head travels around the perimeter, than jumps the gap, creating strands.  I am using a lightweight filament that expands so strands happen regardless of retraction settings.
[image]
 

### Describe the solution you'd like

Add a setting that stops unnecessary travel over an open gap. If it just printed to the end and then reversed direction to print the next row there would be a cleaner edge with no strands. (if such a setting already exists, sorry! I am fairly new to 3D printing and my searches for an answer turned up nothing.)

### Describe alternatives you've considered

I considered printing in different orientations but none would fix the issue. 

### Affected users and/or printers

I believe it would affect every brand of printer.

Asterchades, 2022-02-15:
Combing _kind-of_ does this already. It doesn't so much as reverse the direction, rather it causes travel movements to be restricted (as much as is possible, anyway) based on a user-selected option, such as across walls or within the model hollow. Many profiles already default to having this enabled in some form, and it may give the results you're after so could be worth experimenting with.

On the down side, of course, this does mean the hot nozzle will trace back over filament that has already been laid down, potentially dragging against it and picking up small amounts of it in the event the previous layer was over-extruded or the mechanics of the system have a little slop. Generally this doesn't make much difference to the likes of PLA but it can (and does) have a significant, negative impact on something like PETG.

Reversing direction, in my experience, doesn't always work out as planned. While a line between point X and point Y should be the same laid down in either direction, slight shifts can be introduced depending upon the printer's structure. These shifts can be a nightmare for diagnostics (such as when testing the Arachne betas).

johnnieq2, 2022-02-17:
Thanks. I will take a look for the comb settings in Cura. This is a lightweight PLA so it expands as being printed, which may be a problem. :)

closed 2022-02-17 by johnnieq2