Ultimaker/Cura #11482: Show configuration folder not working Ubuntu

DAC957, 2022-02-15:
### Application Version

4.13.1

### Platform

Ubuntu 20.04.3 LTS

### Printer

Ender 3 v2

### Reproduction steps

selected via menu "help, show configuration folder" 

### Actual results

nothing happens ie no folder opens

### Expected results

Im new to Cura but thinking an file manager window of some kind should have opened.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I cannot figure out how to use above directions to find cura.log in Ubuntu- sorry, Im new to Linux.
Cura is running in compatibility mode presumably because the PC is old and graphics cannot handle Open GL 4.2. Wondering if this could be the problem?

fvrmr, 2022-02-15:
Hi @DAC957 thank you for your report.
Are you using the AppImage?

DAC957, 2022-02-15:
Hello @fvrmr, I am 85% sure I installed via the snap store application bundled with Ubuntu. Store not working at the moment but I found the version at snapcraft.io:   https://snapcraft.io/cura-slicer

fvrmr, 2022-02-17:
Thank you, we don't maintain the snap package of Cura. Please report your issue to the maintainer of that package.
We only maintain the AppImage: https://github.com/Ultimaker/Cura/releases/tag/4.13.1 
I will close this issue since this is probably not a problem with the AppImage.

closed 2022-02-17 by fvrmr