Ultimaker/Cura #11487: ChangeAtoZ is not adjusting retraction distances.

64bittuning, 2022-02-15:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Tronxy XY-2 Pro

### Reproduction steps

Loaded Retraction Tower model from "Part for calibration" plugin.
Enabled 4x "ChangeAtoZ" scripts to alter retraction to 0.2, 0.4, 0.6, and 0.8mm at 40, 80, 120, and 160mm respectively.
Sliced and printed file. Noticed retractions and stringing looked terrible.
Inspected the gcode file and found that no retractions were commanded at all.
Verified retraction was enabled. initial value was set to 0.0. Assumed this was the cause, resliced the file with 0.1mm.
Saved and inspected gcode again. retraction (linear) remains at 0.1mm throughout entire print. Never changes.
Modified "ChangeAtoZ" setting for layer 120 to add retraction speed change. Inspected file and found 

;LAYER:120
;[CAZD:
;RETRACTFEEDRATE 2700.0
;RETRACTLENGTH 0.6
;:CAZD]
G1 F2700.0 E-0.1;[CAZO:G1 F3900 E-0.1:CAZO]

### Actual results

Modified "ChangeAtoZ" setting for layer 120 to add retraction speed change. Inspected file and found 

;LAYER:120
;[CAZD:
;RETRACTFEEDRATE 2700.0
;RETRACTLENGTH 0.6
;:CAZD]
G1 F2700.0 E-0.1;[CAZO:G1 F3900 E-0.1:CAZO]

It seems as though "ChangeAtoZ" is not changing retraction distances.

### Expected results

Layers 40, 80, 120, and 160 and all subsequent lines should have changed to 0.2, 0.4. 0.6. 0.8mm retraction values.

Layer 40 is file line 5839
Layer 80 = 9682
layer 120 = 13525
layer 160 = 1736
[single wall calibration cylinder for retraction bug test.txt](https://github.com/Ultimaker/Cura/files/8073785/single.wall.calibration.cylinder.for.retraction.bug.test.txt)
[individual line search results for retraction showing no change from one tenth.txt](https://github.com/Ultimaker/Cura/files/8073786/individual.line.search.results.for.retraction.showing.no.change.from.one.tenth.txt)
9

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[0 to 8tenths retraction test PETG.gcode.txt](https://github.com/Ultimaker/Cura/files/8073863/0.to.8tenths.retraction.test.PETG.gcode.txt)
[0 to 8tenths retraction test PETG.zip](https://github.com/Ultimaker/Cura/files/8073867/0.to.8tenths.retraction.test.PETG.zip)

fvrmr, 2022-02-18:
Hi @64bittuning thank you for your report.
The Change At Z processing script is in experimental. I checked 4.7.1 (this was introduced in 4.7) with your project file and there I get the same gcode results. Did you noticed this before?

64bittuning, 2022-02-22:
@fvrmr I was running either 4.12 or 4.12.1, I do not recall which. (backstory) I was having a significant jamming issue with my small machine, and loaded up the retraction tower. I did two different tests. 1-5mm and 0.2 to 1mm (its a direct drive machine). The test printed appropriately and I found my best retraction to be 0.2mm with PLA. I had never done a retraction test in this manner before, so I went and did the same test on my large machine. 1-5mm and found acceptable results at 3mm, then ran it again with 3.0-3.8mm. The machine looked to produce the best results at 3.2mm

Being that the test ran properly on two machines prior to this, I would it odd that the failure would present now. This makes me curious to find out if the fault is caused by the Tronxy profiles which I recently updated. I'm going to uninstall my Cura today and do some tests. I will report back with my findings.

64bittuning, 2022-02-23:
@fvrmr It looks like the issue is specifically with choosing to run with "Relative Extrusion" selected. 
I have reset my install back to a base 4.13.1 and deleted all custom profiles, so its as absolutely box stock as possible. I loaded two cylindrical shapes and placed them 20mm apart on the X axis, and set them to a single wall (for code simplicity).
Attached you will find the two gcode files. In the standard file, you can search "CAZO" and find each line where the changes were made and you can see where they increase in 50 layer increments. 50 = 1mm, 100 = 2mm, 150 = 3mm and so fourth.

[image]

[image]
[TXY2PRO_flag sleeve.gcode.txt](https://github.com/Ultimaker/Cura/files/8125171/TXY2PRO_flag.sleeve.gcode.txt)
[TXY2PRO_flag sleeve_relative.gcode.txt](https://github.com/Ultimaker/Cura/files/8125172/TXY2PRO_flag.sleeve_relative.gcode.txt)

Ghostkeeper, 2022-02-26:
It looks like the Change At Z script doesn't handle relative extrusion at all. Not with the retractions, nor with the flow rate adjustments. It was probably never implemented to handle that.

fvrmr, 2022-03-10:
We have discussed this with the team and decided to defer this issue. It is not a priority for us to fix this. I hope you understand.

64bittuning, 2022-03-10:
> We have discussed this with the team and decided to defer this issue. It is not a priority for us to fix this. I hope you understand.

I fully understand. It's not an overwhelmingly large issue, especially after I identified the exact criteria for the failure.
If you don't mind, I'd suggest changing lines 299 - 304 of changeatz.py to 

`                "caz_change_retract": {
                    "label": "Change Retraction",
                    "description": "Indicates you would like to modify retraction properties.  Does not currently work with Relative Extrusion.",
                    "type": "bool",
                    "default_value": false
                },`
                
This should serve an an interim solution until such a time as you (the devs) decide to address the issue.

[image]

msticninja, 2022-07-06:
Would sure love to see this fixed. I had to go back to Simplify3D for all of my prints due to this issue.

novamxd, 2022-08-17:
Hey folks,

I'm the one who did the revamp of the plugin at 5.2.0. So I actually designed it to work with Relative Extrusions and Firmware retractions. The downside, though, is that it needs to know what the starting retract length was set at when the GCODE was synthesized as it needs to make relative adjustments to all follow up commands.

So if you want to do something like this with Linear Move, do the following:
1. Set the retract length in your print settings to anything over zero
2. Use ChangeAtZ to set the retract length to 0 at layer 1
3. Use the other ChangeAtZ as needed
4. ???
5. Profit!

novamxd, 2022-08-17:
By the way, this it is the command it looks for when trying to autodetect the retract length. As you can see here, it's missing in the GCODE with a retract length of 0 but present when I set the retract length to 0.6.

[image]

tstone52, 2022-10-07:
I was about to open an issue for this. I am having this same problem in Cura 5.1.0.
I need to be able to use both relative extrusion AND a post process retraction change, but I cannot due to this bug. I have found that I 4.5mm retract works for me, except I prefer a lower retraction on Layer 0 only. I get better bed adhesion with lower retraction, especially on models with small bed-contact-areas. Therefore I want the first layer retract to be low (I suppose it could even be 0) and every other layer to be 4.5mm.

novamxd, 2022-10-29:
Hey @tstone52 ,

Sorry for the late reply. If you're still having this issue, can you post a screenshot of your ChangeAtZ settings and include your GCODE? I can take a look and see if there's anything that stands out.

> You mentioned firmware retraction. I am not using firmware retraction. Do I need to use firmware retraction for the method you described to work?

No you don't need to, but firmware retraction is more reliable for detection because it uses a specific GCODE command and thus the code doesn't need to do some wizardry to attempt to detect it for you. But not all 3D printers support firmware retraction and AFAIK it's not enabled by default. Looking at the printer settings for Cura 5.2, I actually don't even see the option to enable firmware retractions anymore, so I should probably remove that from the plugin.

novamxd, 2022-10-29:
Ah. The firmware retraction option is part of the [Printer Settings Plugin](https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/PrinterSettingsPlugin)

tstone52, 2022-11-01:
> Hey @tstone52 ,
>
> Sorry for the late reply. If you're still having this issue, can you post a screenshot of your ChangeAtZ settings and include your GCODE? I can take a look and see if there's [...]

Sure thing. I appreciate the reply. Here are a few screenshots and the gcode (as a txt file). 

[image]
[image]
[image]

[AI3M_190C_FlexSnake_RetChangeLayer0_RelExtrudeOn.txt](https://github.com/Ultimaker/Cura/files/9913204/AI3M_190C_FlexSnake_RetChangeLayer0_RelExtrudeOn.txt)