Ultimaker/Cura #11489: I put a tick on "generation of supports" and nothing is generated. only manually.

LEXUS9, 2022-02-15:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

anycubic vyper

### Reproduction steps

I put a tick on "generation of supports" and nothing is generated. only manually.

### Actual results

I put a tick on "generation of supports" and nothing is generated. only manually.

### Expected results

I put a tick on "generation of supports" and nothing is generated. only manually.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I put a tick on "generation of supports" and nothing is generated. only manually
[image]
.

GregValiant, 2022-02-16:
With the model loaded use the "File | Save Project" command to save a 3mf file.  Then zip it up and post it here.

Asterchades, 2022-02-16:
Automatically generated supports don't show on the Prepare tab (the one pictured) - only the preview (middle tab). You have to slice the model after telling it to generate them, then switch to the preview of said slice, before you can see where they're going to be.

LEXUS9, 2022-02-19:
[AV_cat_rogue.zip](https://github.com/Ultimaker/Cura/files/8101988/AV_cat_rogue.zip)

> With the model loaded use the "File | Save Project" command to save a 3mf file. Then zip it up and post it here.

GregValiant, 2022-02-19:
This is with no changes to the 3mf file.  (It's a different model.)
Did you switch to the "Preview" window as @Asterchades suggested?
[image]

LEXUS9, 2022-02-19:
> Automatically generated supports don't show on the Prepare tab (the one pictured) - only the preview (middle tab). You have to slice the model after telling it to generate them, then switch to the preview of said slice, before you can see where they're going to be.

thank you very much. i'm a dumb newbie

closed 2022-02-19 by LEXUS9

Asterchades, 2022-02-19:
We were all beginners at some point. Some of us learn by experimenting, others by asking questions - both are perfectly valid ways to improve.