Ultimaker/Cura #11491: Cylindrical mold + option to fill mold

myearwood1, 2022-02-16:
### Is your feature request related to a problem?

I have a specific need to make a 3d printed item encased in a cylindrical mold much like a binder jet printer makes an item encased in powder. The mold and item would be 2 different materials.

### Describe the solution you'd like

I imagine your mold feature builds a mold around an object. That object is excluded. I want the option to build the mold around the item.

### Describe alternatives you've considered

The mold is empty and I need it to contain the item as well.

### Affected users and/or printers

In the pursuit of ultimate support, if a support material completely encased an item, but could be extracted from the mold, the item would be more like the binder jet items where the powder is the support material.

GregValiant, 2022-02-16:
A picture is worth a thousand words.  This one is mine.  I have no idea if it is even remotely like what you are looking for.

[image]

GregValiant, 2022-02-16:
If the model inside the mold was a drinking glass, then I would make an insert connected to the mold "roof" with locating pins to align the roof to the centerline of the mold.
Cura is a slicer and not a CAD program.  You can put parts together in Cura (kind of) but the process for what you want to do generally takes place outside Cura.  For a more typical mold, the sprues, vents, and locators need to be added in CAD.

myearwood1, 2022-02-16:
That's a good point, but my interest is for a binder-jet like approach. The
powder and binder will be produced by an inkjet simultaneously.

GregValiant, 2022-02-17:
If the inner mold will be resting on the binder stuff then you might have an issue with the "first layer" of the inner mold core.
This is a section of a drinking glass within a mold.  The arrow is pointing to what I see as a problem area.  Getting it apart might  also be an issue.  I'm thinking it would need a release agent.  The inner mold core could be hollow.

At any rate, this conversation goes beyond the scope of what Cura does.  But to print the model as shown below would not be a problem as long as the red "floor" of the inside of the glass is solid before the first layer of the mold core material goes down.
[image]

MariskaMaas, 2022-02-17:
Hey, in my spare time I like making silicone molds with FFF. 

I have to admit that I don't understand this specific assignment, but.....
I have a standard positive mold on Thingiverse that might help you out. 
It's not cylindrical but spherical but you might be able to fill it with the material you need + the object you need.
https://www.thingiverse.com/thing:3795542/files

[image]
.

myearwood1, 2022-02-17:
I would love to explain. I intended to fill a container with the both the
mold material and the item material layer by layer.

To that end, I am trying to get the mold made as support material around
the item.

myearwood1, 2022-02-17:
The physical materials are in powder form, so no release agents required.

There is not even any binder. It is being assembled like 2 different kinds
and colors of wet sand to make a sand castle encased in a sand cylinder.

It is a new invention for 3d printing.

fvrmr, 2022-02-18:
Could you explain a bit more why you would like to this option for a mold in Cura. 
What would be the benefit and for what things would you want to use this?

myearwood1, 2022-02-18:
I am creating a hybrid 3d printer that uses one material as support and
another material as the item.

Imagine graphite and glass. The entire cylinder will be printed inside a
crucible and then put in a kiln.

There is another application but I'm not sure I can say.

fvrmr, 2022-02-25:
Would a cutting mesh help you? https://support.ultimaker.com/hc/en-us/articles/360013370140-How-to-adjust-print-settings-of-a-part-of-my-model-in-Ultimaker-Cura

myearwood1, 2022-02-25:
Hi!

Perhaps!!

These are a mesh to surround the object and also one to infill the object?

They can be printed along with the object?

Ghostkeeper, 2022-03-07:
Cura already has a mould feature. It creates a hollow mold though, printing just the mould instead of the object. But since the mould is a per-object setting, you can simply load your model in twice, set one of them to print as a mould and with the other extruder, and then merge them. One of your models then just gets printed normally, while the other becomes the mould around it.

[image]

myearwood1, 2022-03-07:
Excellent! One last trick then. Once merged there will be 2 different
materials per layer. Can Cura output each layer as a 2 color image?

myearwood1, 2022-03-07:
Can Cura be set up with a cylindrical build area? I think so, but please
confirm. The slicing to produce separate images would help with the binder
jetting technologies.

fvrmr, 2022-03-17:
You can set the build plate shape to elliptic in the machine settings.

fieldOfView, 2022-03-17:
> Can Cura output each layer as a 2 color image?

No, Cura does not export per-layer images. Cura only exports sliced data as gcode toolpaths.

myearwood1, 2022-10-11:
Then I request you add the feature to output layers as multi-color images.
I was patent pending on a multi-color resin based printer. This would allow
Cura to be used in binder jetting inkjet machines.