Ultimaker/Cura #11496: ### Welcome to the discussion thread of the second beta of Arachne!

gguerif, 2022-02-16:
### Welcome to the discussion thread of the second beta of Arachne!
Earlier this year, we released the first beta of Ultimaker Cura Arachne engine. Which received an overwhelmingly positive response, thank you for all your feedback! 

You are here because you probably want to share some feedback about Arachne engine beta 2! Share your feedback, show your results and/or provide us your input for print profiles for your preferred 3D printer with us here in the comments below.

We really appreciate it if you would follow the following guidelines while posting:

• Read the [release notes](https://github.com/Ultimaker/Cura/blob/libArachne_rebased/resources/texts/change_log.txt) before posting. Make sure that it isn’t already a known issue.
• Please note the printer you used, as well as any profiles or changes.
• If you need to share a model, please save as a project (file – save project) and attach here as a .zip.
• A picture really does say more than a bunch of words sometimes.

**New features:**
-  Back pressure compensation
-  Printing order filling gaps
-  Print quality enhancements

**New settings:**
- Split Middle Line Threshold
- Add Middle Line Threshold
- Wall Distribution Count, which is only enabled/visible if the Inward Distributed strategy has been selected

**Bug fix highlights:**
- Fixed a bug with concentric skin lines
- Fixed a bug with opening old project files in Arachne
- Fixed a bug where the width was not adjusted
- Fixed a bug with Line Compactness behaviour was inversed for Central Deviation
- Fixed a bug where blobs in g-code were generated
- Fixed a bug with Minimum Wall Line Width setting was used if Print Thin Walls was disabled
- Fixed a bug where the Inner Line Width was wider than the end of the model
- Fixed a bug where the Seam placement was not on the correct corner
- Fixed a bug where the centerline of Central Deviation is not generated
- Fixed a bug with Bottom Skin Expand Distance at zero caused skin to expand beyond walls

**Known issues:**
- Gaps in inner wall where it has enough space to print one line
- Unwanted travel moves when monotonic order in enabled
- Printer pauses at certain layer creating dripping spots
- Retracted unnecessary travel moves

[Download the beta now!](https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_beta_2)

_Originally posted by @fvrmr in https://github.com/Ultimaker/Cura/discussions/10931_

gguerif, 2022-02-16:
Hello, on arachne beta 2  I can enter my own start & end gcodes (for a purge line) but they don't stay on.
Am I alone in this case ?Thanks

closed 2022-02-16 by gguerif

fvrmr, 2022-02-22:
Hi @gguerif thank you for your report. 
This looks the same as https://github.com/Ultimaker/Cura/issues/11214
Could you share your log file? https://github.com/Ultimaker/Cura#logging-issues