Ultimaker/Cura #11497: For dual extrusion: add the possibilty to select which tool starts first

Keybored02, 2022-02-16:
### Is your feature request related to a problem?

I oftentimes find myself printing with a multimaterial (or multicolor) setup. I usually adopt the "assign the toolhead and merge the two pieces" approach for most printjobs. The issue is the consistency. 
When I merge the exact same two models a second time (even without changing settings) and slice again, the new first layer differs from the previous one. If Extruder 1 started first then, now is Extruder 2 to start the print (and viceversa). The assignment seems random, but I suspect it has to do with the merging itself and some microscopic mesh misalignments.
This might sound trivial, but for many models it's quite invalidating. The bottom quality relies heavily on which part is printed first, especially for lettering and small 2D features. It's quite important for specific couples of materials, especially if the melting points are really far part.
Found on Cura 4.9 and later. Did not chek previous versions.

### Describe the solution you'd like

Since there is a hierarchy that decides with toolhead goes first, an override button under the "Dual Extrusion" menu would probaly be everything it's needed.

### Describe alternatives you've considered

The only solution I found is to assign a skirt or brim to one of the two extruders, as doing such will make the first one to extrude by default. This of course does not apply on a machine with 3+ extruders.

### Affected users and/or printers

Pretty much every printer with more than a single extruder, and jobs that require both extruder to be used on the first layer.

### Additional information & file uploads

I have prepared 4 examples:

Example 1 and 2 are sliced from the same merge, with the model from Example 2 being translated in the plane. 

[Example_3_PLA_0h_21min.txt](https://github.com/Ultimaker/Cura/files/8084266/Example_3_PLA_0h_21min.txt)
[Example_4_(skirt_included)_PLA_0h_24min.txt](https://github.com/Ultimaker/Cura/files/8084267/Example_4_.skirt_included._PLA_0h_24min.txt)
[Example_2_PLA_0h_23min.txt](https://github.com/Ultimaker/Cura/files/8084265/Example_2_PLA_0h_23min.txt)
[Example_1_PLA_0h_23min.txt](https://github.com/Ultimaker/Cura/files/8084264/Example_1_PLA_0h_23min.txt)

Example 3 is the same two models being re-merged again: notice how the first extruder changes.
Example 4 is Example 3 plus a skirt, to prove the effectiveness of my "solution".

I am sorry for the .txt, I didn't want to send a .zip.

fvrmr, 2022-02-18:
Hi @Keybored02 thank you for your request.
Could you also share a project file (file - save project) instead of the gcode files? If you don't want to send a .zip you can also change the .3mf in a .txt.

Keybored02, 2022-02-19:
I don't know how much the project file will help, since I encounter the differences after re-merging the same two files in the same project. Each slice is consistent within the same merge, but differs from merge to merge (if that makes sense). But I'll upload it anyways:
[Chip_1_PLA_0h_29min.txt](https://github.com/Ultimaker/Cura/files/8101955/Chip_1_PLA_0h_29min.txt)

To further explain the impact that it has on quality, here are two different merges of the same models. The first one follows the order extruder 1 (black) --> extruder 2 (silver), with the outline of the text being printed first. The second is the opposite, with the silver text itself being printed first. It definitely has an impact.

[image]

fvrmr, 2022-02-21:
Thank you, your picture explains your problem! I will bring it up for discussion with the team. Keep you posted!

fvrmr, 2022-02-22:
I have discussed it with the team. And we decided to defer this. 
A work around is for example to select the Extruder 2 as `Build Plate Adhesion Extruder`. When selecting extruder 1, extruder 1 will start. I hope this will help you!
[image]

Keybored02, 2022-02-22:
Thanks for keeping me posted. I'm already using this workaround with success. My concern was with more than 2 extruders, like with a Geeetech A10T or similar printer. I haven't found a solution for that yet. But thanks for considering the problem!

closed 2023-07-13 by github-actions[bot]