Ultimaker/Cura #11498: Update adding decimal after temperature

ColonelFitzgerald, 2022-02-17:
### Application Version

4.13.0

### Platform

Mac Monterey 

### Printer

Guider 2s

### Reproduction steps

With the new update, Cura is now adding a decimal after the initial bed and extruder temperature in the gcode. When I then send this to my Guider 2s, it seems to not be able to read the temperature setting and sets the bed to 50c and nozzle to 180c.

Previously this hasn't been an issue, and when I looked at older successful prints the temperature is a whole number in the gcode

I tried uploading the files through PolarCloud and USB (files sliced in Cura and opened in Flashprint no longer read correctly) and had the same issue.

### Actual results

Temperature set correctly once I had manually deleted the .0 in the gcode

Previous successful gcode:
;Generated with Cura_SteamEngine 4.12.0
M82 ;absolute extrusion mode
;START CODE
M104 S220 ;set initial extruder temp
M140 S65 ; set initial bed temp
;M116 ; wait for set temps
;M105 ; get extruder temp
M82 ;absolute extrusion mode

Error Code:
;Generated with Cura_SteamEngine 4.13.0
M82 ;absolute extrusion mode
;START CODE
M104 S255.0 T0;set initial extruder temp
M140 S115.0 T0; set initial bed temp
;M116 ; wait for set temps
;M105 ; get extruder temp
M82 ;absolute extrusion mode

Fixed manually successful code:
;Generated with Cura_SteamEngine 4.13.0
M82 ;absolute extrusion mode
;START CODE
M104 S255 T0;set initial extruder temp
M140 S115 T0; set initial bed temp
;M116 ; wait for set temps
;M105 ; get extruder temp

### Expected results

After I manually removed the .0 the temperature ran correctly on the printer

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Is it possible to automatically get rid of the .0 after the temperature?

fvrmr, 2022-02-17:
Hi @ColonelFitzgerald thank you for your report.
I could not reproduce it, could you maybe share a project file (file - save project) and share it as a .zip here?

ColonelFitzgerald, 2022-02-18:
I've attached both the project file and the gcode it creates:

[2.18_Offset_ hand.3mf.zip](https://github.com/Ultimaker/Cura/files/8092994/2.18_Offset_.hand.3mf.zip)

[2.18_Offset_ hand.gcode.zip](https://github.com/Ultimaker/Cura/files/8093022/2.18_Offset_.hand.gcode.zip)

GregValiant, 2022-02-18:
Yep.  This is with 4.13.1:
If I enter the temperature as 260.0 then it displays in Cura as 260.0 and shows up in the Gcode as 260.0.
If I enter the temperature as an integer 260 then it displays in Cura as 260.0 and shows up in the Gcode as 260.

M104 S259 T0            ;Entered as "259" and displayed as "259.0"
M140 S115.123 T0     ;Entered as 115.123 and displayed as "115.123".  @fvrmr -  **Why is there a T0 in this line?**

My Creality printer (Marlin) ignores the decimal and appears to truncate the temperature numbers to integers.

ColonelFitzgerald, 2022-02-21:
The T0 I added recently in an attempt to fix the issue, but it doesn't seem to have an effect.

I am still finding the .0 is causing issues

GregValiant, 2022-02-21:
This is from fdmprinter.def.json.  All the main temperature textboxes are defined much the same.  I checked back to Cura 4.7 and they were defined as "float" there as well.  (The "minimum_value" of "-273.15" seems a bit cool for printing.)

"material_print_temperature":
                {
                    "label": "Printing Temperature",
                    "description": "The temperature used for printing.",
                    "unit": "°C",
                    **"type": "float"**,
                    "default_value": 210,
                    "value": "default_material_print_temperature",
                    "minimum_value": "-273.15",
                    "minimum_value_warning": "0",
                    "maximum_value_warning": "285",
                    "maximum_value": "365",
                    "enabled": "machine_nozzle_temp_enabled and not (material_flow_dependent_temperature)",
                    "settable_per_mesh": false,
                    "settable_per_extruder": true
                },
@ColonelFitzgerald  You can manually change the: ["type": "**float**"] designations to ["type: "**int**"] on the temperature textbox definitions.  That should work for you as at least a temporary fix.  If you can only enter an integer then it appears that Cura will put that integer into the gcode.
Be advised that a hack like that may may cause breakage down the road somewhere.  Make a backup copy of the file before altering anything.  @fvrmr or another team member would know better and may advise against changing that as well.

What firmware is your printer running?  (My Marlin firmware ignores this and I would assume that the UM printers also don't care.)

fvrmr, 2022-02-22:
I checked with the team and changing the type: float to type: int could work. But it seems like the cause of adding a decimal has to be something else, since type: float has always been in the fdmprinter.def.json
We need to do a bit more investigation. Keep you posted.