Ultimaker/Cura #11499: More sophisticated seam control for big nozzles.

karabas2011, 2022-02-17:
### Is your feature request related to a problem?

I faced with the fact that it is impossible to hide seam when using big nozzle diameters ( starting from 0.8 I think).
The bigger diameter the worse result. Because:
Big nozzle tends to drag extruded plastic line when starts to travel. That creates holes. 
I prefer bulbs because  they can be removed with trimmer (or smth ). Filling holes is a pain.

### Describe the solution you'd like

So I suggest two possible options.
1 negative retract - extrude plastic when start to travel  insteed of sucking  (allowing negative retract values)
2 overlap perimeter ends

### Describe alternatives you've considered

Maybe  dont stop extruding at all when seaming.

### Affected users and/or printers

It will affect all big nozzle users.

fvrmr, 2022-02-18:
Hi @karabas2011 thank you for your feature request. 
We are aware of this issue and have researchers looking into improving z seams. You can place the seam to hide it, but this depends on your model. It might not have an effect on cylindrical models.

fieldOfView, 2022-02-18:
This article from the Settings Guide plugin has a number of strategies to help hiding the seam (including ways to overlap perimeter ends):
https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/troubleshooting/seam.md#closing-the-seam

karabas2011, 2022-02-24:
I looked into other slicers to hide seam. Most of them affects only external  perimeter. But internals are important too in many cases. So I wonder why no option to internal perimeter?

pixie-wrangler-irl, 2022-02-26:
Both of your requested items already exist in some form.  There is "Retraction Extra Prime Amount" and "coasting" or "wipe distance".