Ultimaker/Cura #11500: Filter settings based on user change.

SergiuToporjinschi, 2022-02-17:
### Is your feature request related to a problem?

I usually look for changes that I made in settings.
If you are new in 3d printing, and not only. For example when I want to tune some setting I usually set a value for a setting and then print it come back to the same setting and change it again and print again.

### Describe the solution you'd like

In settings add another default filter, like those that are already there like Basic, Advanced, Expert, All, with Modified and this should show only what is changed by user, only those that have Restore button.

### Describe alternatives you've considered

Create multiple custom filters

### Affected users and/or printers

not the case

### Additional information & file uploads

[image]

fvrmr, 2022-02-18:
Hi @SergiuToporjinschi thank you for your feature request.
We have a similar request see: https://github.com/Ultimaker/Cura/issues/11017 where it will show where changed settings are. 
Also we have this request to show favorite settings: https://github.com/Ultimaker/Cura/issues/10352 
Both have a ticket on our backlog. Do you think this would help you?

SergiuToporjinschi, 2022-02-21:
Hi @fvrmr,
Hmm yes it helps a bit, my scenario is different, because in Cura there are a lot of settings that you  just need to tune-up for compensating some printer problems for example "horizontal expansion". How am I tuning this. 
For example I want to print something very precise and I have know printer problem (For sure I will fix it later) but I'm un a hurry and I need  that printed.
I will start a print with initial value. 
Measure it and then go to Cura and decrease the value.
Print it again
Measure again;
Tune the value once again;
Each time I need to find for that setting in a big  pile of settings. 
For example brim distance. The same I cannot say for sure what value should be there because some times depends on temperature because if I print with a temp slightly bigger the PLA will be more fluid so in this case I have coupe of settings that I need to tune at once.

In the end is your decision about how you will make this process easier. For sure those to tickets seems to help a bit, not perfect for me :) but you will have to decide what and how you are make this better.
or Maybe some plugins can help to satisfy everybody :) 
I also looked in this direction to create a plugin but I could not find too much info about how to set-up a project to build this plugin and I think it needs C# !??! maybe!??! and that is not my favorite language :)

Thank you!
Have a nice day!
Sergiu Toporjinschi

fvrmr, 2022-02-21:
Thank you for your reply, I will bring it up with the team. Keep you posted!

fvrmr, 2022-02-22:
I have discussed it with the team and we decided to add you request to the existing ticket on our backlog as a solution.

gonzalo-massa, 2022-05-15:
I really like this idea and I think I'll be better in combination with #11017. With "combination" I mean to do both, but also if there will be an icon or mark to indicate which sections have modified settings, maybe clicking that icon should filter the settings to show the modified ones.

garyzz, 2025-03-18:
Did this ever get implemented?