Ultimaker/Cura #11502: Missing Save Function in Printer machine settings

bebr42, 2022-02-17:
### Application Version

4.13.0

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

1. Change printer "machine Settings", like Start G-code
2. 'Save' Button is _missing_

### Actual results

Printer settings are some saved permanently after 'Close', sometimes not. 

### Expected results

Save Button that does what it says.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

The idea of the developers seems to be that any changes to the settings are saved only when a new printer is setup.
This is intransparent to the user and yields likely irritating results. 
See also:
https://community.ultimaker.com/topic/30585-cura-not-retaining-custom-start-gcode-in-machine-settings/

fvrmr, 2022-02-18:
Hi @bebr42 thank you for your report.
I would see this more as a new feature. Do you want an extra button with save or replacing the close with save?

bebr42, 2022-02-18:
IMHO it is a real bug because saving settings is a _basic_ feature that every user would expect. 

Yes, [Save] instead of [Close] might make sense.
But just renaming the button will not suffice...

fvrmr, 2022-02-22:
I have discussed it with the team and in a new design the close button is removed. 
Adding a save/cancel button is something we do not want to add, since it is a bit complex and just closing should save your work. Only you can not revert your changes. I hope you understand.

closed 2022-02-22 by fvrmr

bebr42, 2022-02-23:
Sorry, there seems to be a misunderstanding. 
The button is just one possible solution - the actual problem is a missing "save" function for printer settings.
Is the developer team really stating that these settings should **never** be changed?

fvrmr, 2022-02-24:
Sorry I didn't explain it correctly then. But in the next release the close button is gone. 
And if you want to add a save function you also need a cancel function so you could undo your settings. And that is something we don't want.

However closing the window should save the settings. And the X at the right top in that screen is enough, so that is why the close will be gone. 
Does that make sense to you?

bebr42, 2022-02-24:
I had the experience that my 'machine setting' edits did not persist. 
So sorry for letting this bug slide to the question of buttons or UI solutions. 
So I opened a [new Feature request here](https://github.com/Ultimaker/Cura/issues/11533).