Ultimaker/Cura #11507: Failure when changing printers

ljohnsonnsb, 2022-02-18:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Custom Idex 

### Reproduction steps

After using  the Idex in duplication or mirror mode and then switch to another printer Cura crashes. 

### Actual results

Cura crashes and generates this error:
Traceback (most recent call last):
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\Settings\ExtruderManager.py", line 425, in getExtruderHasQualityForMaterial
KeyError: 'empty_material'

I send the report. 
After repeating these steps 3 or 4 times it actually changes. 
NOTE: If it right click on the model in Idex mode it will show 'Extruder 1 - empty ' and 'Extruder 2 - empty' 

### Expected results

I would expect it to switch to the new printer. 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

LOG FILE
[cura.log](https://github.com/Ultimaker/Cura/files/8098827/cura.log)

fvrmr, 2022-02-22:
Hi @ljohnsonnsb thank you for your report.
I am not able to reproduce this issue.
Could you share your configuration folder of 4.13 with me? You can find that folder here: `C:\Users\<your username>\AppData\Roaming\cura`

fvrmr, 2022-02-24:
Hi @ljohnsonnsb thank you, I can't access that link. Can you check that if you shared it with everyone?

gershb0mb, 2022-03-05:
Hi @fvrmr, I have this exact same issue when changing from a dual extruder configuration to a single extruder; traceback is the same as OP and after 3-4 crashes Cura will open with the printer that I was trying to switch to. There is no issue going from a single extruder configuration to a dual extruder configuration. My configuration file is attached.
[4.13.zip](https://github.com/Ultimaker/Cura/files/8191450/4.13.zip)

fvrmr, 2022-03-16:
@gershb0mb thank you for your configuration folder. I can reproduce this and will bring it up for discussion. 
Keep you posted.

fvrmr, 2022-03-21:
We have created a ticket on our backlog for this. Devs see CURA-9059