Ultimaker/Cura #11508: Extra skin wall speed should correspond top surface skin speed at top surface layer

Nikolay-Po, 2022-02-18:
### Is your feature request related to a problem?

In Cura 4.13.1 the speed of printing of Extra Skin Wall is corresponding to Top/Bottom speed which is annoying. If I'm printing Bottom and Top at high speed but want to finalize top surface at low speed, extra skin wall lines are keeping high speed of Top/Bottom which is not I want.
For example, while printing initial layer, Extra Skin printing speed is corresponding initial layer printing speed. See a screenshots. The first is Initial Layer (correct speed). Second is Top Surface layer (incorrect speed for extra skin wall).
[image]
[image]
.

### Describe the solution you'd like

Add a checkbox for applying "Top Surface Skin Speed" at last top layer for each type of lines. For example, In "Speed" category, after "Top Surface Skin Speed" parameter, add a checkbox: "Use Top Surface Skin Speed for all" with a description: "Top Surface Skin Speed will be used for any line types of top surface layer when checked".

### Describe alternatives you've considered

Or just unify the speeds of top surface layer to one common speed value "Top Surface Skin Speed". Instead of printing Outer Wall, Extra Skin Wall and Top Surface at different speeds, print all lines of top surface at one common speed "Top Surface Skin Speed".

### Affected users and/or printers

Only users who used Extra Skin Walls are affected. Any model of FDM printers are affected.

### Additional information & file uploads

Example Cura project is attached (corresponding screenshots are attached to problem description section above).
[Extra_Skin_Wall_Speed_Mismatch.zip](https://github.com/Ultimaker/Cura/files/8100274/Extra_Skin_Wall_Speed_Mismatch.zip)
.

fvrmr, 2022-02-21:
Hi @Nikolay-Po thank you for your request.
I will bring it up with the team. Keep you posted!

Ghostkeeper, 2022-03-07:
See also https://github.com/Ultimaker/Cura/issues/7886 .

Back then the issue was recognised, but not considered to have priority. I don't know what the team will answer this time, but perhaps this should be closed as a duplicate to consolidate the discussion to one thread?

Nikolay-Po, 2022-03-07:
Sure, the same thing as described in [#7886](https://github.com/Ultimaker/Cura/issues/7886). But mine ticket contains more clear propositions. Can it be evaluated again and, probably, processed but not deferred?
Extra wall is designated for improving quality of top surfaces. But with a lack of speed control, extra wall become useless in cases when both performance and quality are required.