Ultimaker/Cura #11509: Plugin SentryLogger Error

zawier-git, 2022-02-19:
### Application Version

4.13.1

### Platform

Linux Garuda (5.16.10-zen1-1-zen #1 ZEN SMP PREEMPT Wed, 16 Feb 2022 19:35:21 +0000 x86_64 GNU/Linux)

### Printer

Cr10 S4

### Reproduction steps

Hi,
After running Cura  
[image]

I see information like bellow:
[image]

Then when I remove it using button:
[image]

### Actual results

And after shutting down Cura and opening it one more time I still see the same problem with plugin.
Please , could you write a fix for that?

### Expected results

I was expecting that plugin will be removed and error message be gone. 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Here is my log file:
[cura.log](https://github.com/Ultimaker/Cura/files/8102310/cura.log)

fieldOfView, 2022-02-19:
Use the AppImage provided by Ultimaker instead of the package you use now. Whoever built that package forgot to include required dependencies.

zawier-git, 2022-02-19:
Ok, thank you for information.
Is there any instruction how to install it properly from AppImage? 
My attempt failed and error message was lead me into https://github.com/AppImage/AppImageKit/wiki/FUSE where I wasn't able to figure it out how to handle that on Garuda Linux (Arch based distro).

fieldOfView, 2022-02-22:
The idea is you download the file, make it executable and double click the AppImage.

fieldOfView, 2022-02-22:
You may want to read both answers here:
https://superuser.com/questions/1306621/install-appimage-under-arch-linux

Ghostkeeper, 2022-03-07:
We don't maintain the Arch package. Please report this issue with the Arch package maintainers instead of here. We only support the AppImage. Arch Linux should support AppImages out of the box (as it only requires Fuse). I'm not aware what the changes would be for your flavour of Arch. Perhaps the suggestions here help as well: https://superuser.com/questions/1306621/install-appimage-under-arch-linux . We'll close this issue since it's a problem outside of our control.

The reason why you're getting the error is because even though Uranium doesn't create the PluginObject any more when the plug-in is disabled, it still registers the plug-in. To be able to provide the PluginObject, the register function imports the SentryLogger module, which has `sentry_sdk` in its imports (which it may assume exists in its runtime environment).

closed 2022-03-07 by Ghostkeeper