Ultimaker/Cura #11510: Horizontal scroll bar and nozzle animation missing

zawier-git, 2022-02-19:
### Application Version

4.13.1

### Platform

Linux Garuda (5.16.10-zen1-1-zen #1 ZEN SMP PREEMPT Wed, 16 Feb 2022 19:35:21 +0000 x86_64 GNU/Linux)

### Printer

CR10 S4

### Reproduction steps

After slicing a model and switching into Preview tab I'm able to see only vertical scroll bar, horizontal one is missing:
[image]

### Actual results

Horizontal scrollbar and print head missing. 

### Expected results

As in wersion at windows I was hopping to find similar functionality.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Here is my log file:
[cura.log](https://github.com/Ultimaker/Cura/files/8102320/cura.log)

fieldOfView, 2022-02-19:
Your instance of Cura is running in "compatibility mode", either because you have selected the option "Force layer view compatibility mode" in the General preferences or because your GPU (or driver) does not support OpenGL 4.1, which is required for the full layer view functionality.

zawier-git, 2022-02-19:
I have checked and you were right regarding option "Force layer view.." I have turned it off. My OpenGL version is 4.6 according to mesa package page. Unfortunatelly a problem still occurs.

fvrmr, 2022-02-22:
Hi @zawier-git thank you for your report.
Looking in your log I see OpenGL 3.1 so that is why you see the layer view in compatibility mode. 

```
2022-02-19 15:07:12,528 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [112]: Initialized OpenGL subsystems.
2022-02-19 15:07:12,528 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [113]: OpenGL Version:  3.1 Mesa 21.3.6
2022-02-19 15:07:12,529 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [114]: OpenGL Vendor:   X.Org
2022-02-19 15:07:12,529 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [115]: OpenGL Renderer: AMD REDWOOD (DRM 2.50.0 / 5.16.10-zen1-1-zen, LLVM 13.0.1)
2022-02-19 15:07:12,529 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [116]: GLSL Version:    1.40.0
```

Ghostkeeper, 2022-03-07:
Mesa is reporting to Cura that it supports up to version 3.1. While the package itself may support up to 4.6, it will be limited by what the hardware supports as well.

I think there is really not a lot that Cura can do to get OpenGL 4 to work on your computer.