Ultimaker/Cura #11511: Input shaper

zawier-git, 2022-02-19:
### Is your feature request related to a problem?

As all you great know a Klipper has awesome feature called Input shaper, which isn't available unfortunatelly on Marlin firmware.  This feature is described as game changer during printing because it reduce ghosting really well. 
Here is somple link what can it achive. 
https://www.youtube.com/watch?v=OoWQUcFimX8&list=PLCwJVCYzMkUoj6O6onpVjrb9yxUvjeiDx&index=306

### Describe the solution you'd like

It would be great to have a such functionality in Cura, where slicer connected with a printer could do some vibration measurment and based on that slice a model :) 

### Describe alternatives you've considered

I do realize it's hard to implement, I really do! But I don't see any alternatives for people who would like to stick with Marlin. Unfortunatelly Marlin community doesn't seam to be intrested into developing a new functionality of input shaper. 

### Affected users and/or printers

It will affect everyone who would be equiped in cheap (1-3$) tiny gyroscope. As a finall print result as we saw on Klippers printers were outstanding only by using that calculations. 

Dear Ultimaker team - if you won't implement it, then I'm affraid nobody will support us with that great approach! Please go ahead with that approach ;)

Asterchades, 2022-02-22:
I'm fairly sure this has to be done as part of the kinematic calculations, which are done at the firmware level. Since Cura only has access to the GCode it can't reasonably duplicate the results. It's much the same reason why Linear/Pressure Advance and S-curve cannot be integrated at the slicer level.

Ghostkeeper, 2022-03-07:
Indeed, it is part of the firmware for several good reasons:

* The calibration procedure requires measurement data. There is no way to send that measurement data from the printer to Cura. No g-code standard gives that information. It can't be given in real-time (too slow) so it'd have to be stored on file and given afterwards, but file systems are a whole can of worms on firmware across the market.
* Input shaping also requires a lot of knowledge about the printer hardware that Cura doesn't have. Firmware has intimate knowledge of the printer. Cura only has a few settings and most often their values are quite wrong when it comes to printer characteristics (unless someone has taken the time to research it and fill them all in, but often people don't even know which machine settings exist).
* To get paths with enough resolution to do proper input shaping, Cura would need to generate highly detailed g-code, with close approximations of smooth curves. If not, the sharp corners will degrade the print quality. This is not a problem for Cura's internals and g-code, but the printer CPUs tend to have problems with processing high resolution, not being able to parse the g-code fast enough.

I think it's safe to say that this is out of scope for Cura, and in the realm of firmware only. It's pretty much impossible to achieve.

closed 2022-03-07 by Ghostkeeper