Ultimaker/Cura #11512: does not have ender 3 s1 in printer list

sophialot, 2022-02-19:
### Application Version

4

### Platform

mac

### Printer

ender 3 s1 3d

### Reproduction steps

cant slice and prepare file for printer

### Actual results

does not have ender 3 s1 in printer list

### Expected results

can not slice

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

add this model to printer list

Asterchades, 2022-02-19:
All 3rd party profiles are maintained (including being added) by 3rd parties. The Cura team is only responsible for the Ultimaker profiles.

Any of the existing Ender 3 profiles should work fine with the S1 - as would most, if not all of the other profiles if you take the time to update the bed and filament dimensions. Naturally retraction will need to be adjusted, but the material profiles (which contain the retraction values) are always separate to the printer profile and would need to be tuned anyway.

fvrmr, 2022-02-21:
Closing this issue, since @Asterchades is right, we don't maintain these profiles. 
As you can see somebody already created a pull request with these profiles!

closed 2022-02-21 by fvrmr

nallath, 2022-07-19:
The PR has been merged. The profile is available for 5.2

commit 6ecb99bc (2022-07-19): Add Creality Ender 3 S1 profile - fixes #11512