Ultimaker/Cura #11514: Slicing failed

sdbruder, 2022-02-19:
### Application Version

4.13.1

### Platform

Windows 11

### Printer

Creality Ender 3 S1

### Reproduction steps

trying to print a thin and big wing with 99999999 top layers

### Actual results

slice failed.

### Expected results

successful slice

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[not_slicing.zip](https://github.com/Ultimaker/Cura/files/8103054/not_slicing.zip)

GregValiant, 2022-02-20:
When a setting box turns orange Cura is giving a warning.  You kind of mis-stated the number.  The 3mf file shows you asking for 9,999,999,999 top layers.  That may be too many digits for a variable in the program.  The model slices with 1087 layers.  Try using a number that is closer to reality.
999,999,999 slices.  9,999,999,999 does not.  That extra 9 billion layers seems to be the issue.

fvrmr, 2022-02-21:
Hi @sdbruder thank you for your report.
@GregValiant is correct that the top layers are causing this. 
I will bring this up with the team. Keep you posted.

closed 2022-02-22 by nallath

sdbruder, 2022-02-22:
thanks for the fix, that will not allow stupid values like the one I used.

Thanks!

nallath, 2022-02-22:
Well, in your defense, it also doesn't break on all operating systems.

commit b1f2f6af (2022-02-22): Add max values for int settings that didn't have them yet

Fixes #11514
See also c8b491e9e7915b3e5534dee0d4576d7a1bafca20