Ultimaker/Cura #11516: [Feature Request] Add Cylindrical Gyroid Infill

N-McKinley, 2022-02-20:
### Is your feature request related to a problem?

In designing exposed infill jars for desiccant (to put in the hubs of reels that are in sealed bags) I have found that printing exposed gyroid infill is inefficient when printing cylindrical, conical, and concave/convex shapes. Cylindrical gyroid would provide superior deformation resistance for cylinders et al as well as increasing print speed substantially.

### Describe the solution you'd like

Add a Cylindrical Gyroid Infill option. This would allow much faster printing of hollow cylinders, cones, etcetera; as well as increasing deformation resistance from lateral forces. This would decrease travel moves and stringing on the infill as well, considering the infill path would be essentially one continuous line throughout the layer (in a single hollow cylinder). It would also allow for interesting spring type actions if a solid cylinder is printed in TPU or similar flexible filament.

### Describe alternatives you've considered

printing flat gyroid panes that can be heated, shaped, and glued in place, but that's a lot of post processing and isn't practical for many applications, any frame, rim or additional pieces would have to be printed separately and adhered which is less than ideal.

### Affected users and/or printers

Most users and FDM printers would benefit from the Cylindrical Gyroid Infill option, round objects are common and exposed infill is becoming more common, "solid" springs, impact absorbing feet, filter elements/sieves, lamp shades, light covers... it's late but that's just 30 seconds of brainstorming.

### Additional information & file uploads

https://www.sciencedirect.com/science/article/pii/S0264127519307786?fbclid=IwAR1RMrEorDhyogjWE2xCPEDCK9IH3NOjBFTigVnz9I2vTEW2c6gz7u26--o

fvrmr, 2022-02-21:
Hi @N-McKinley thank you for your request.
I will bring this up with the team. Keep you posted.

N-McKinley, 2022-02-21:
Thanks! Wouldn't that be exciting if it was added.

  (Issue #11516)

fvrmr, 2022-02-22:
I have discussed it with the team and we decided to defer this request.
This would only affect one type of a model, like conical or cylindrical shapes, so that is why it will not get a high priority for us to implement. You or anyone else is free to create a pull request.

closed 2023-07-13 by github-actions[bot]