Ultimaker/Cura #11517: More ironing setting

delious87, 2022-02-20:
### Is your feature request related to a problem?

A thin layer on the model can be damaged by ironing. There are only two options allowed - all surfaces and only highest. 

### Describe the solution you'd like

Need some additional settings like "not ironing on thin site of the model".

### Describe alternatives you've considered

Ensure all layers are thick enough or mb add some by cura.

### Affected users and/or printers

Quality of prints

fvrmr, 2022-02-22:
Hi @delious87 thank you for your request.
We have a setting ironing inset and this could also help that it will not iron on thin parts of your model.
Read here more about this setting: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/top_bottom/ironing_inset.md
Could that work for you?

delious87, 2022-02-25:
> Could you give me an example/screenshot for what kind of models you would like this.

https://www.thingiverse.com/thing:1183353/files
Base_fixed.stl base for window stopper
There are some thin plates destroyed by ironing.

Ghostkeeper, 2022-03-12:
We can't make filtering-type settings that select on all the types of features of a 3D model you might want to iron or not iron. There's too many cases.

This is why we have cutting meshes. Best use those.

Perhaps a better approach for this sort of thing would be an easier way to select zones for ironing than a cutting mesh.