Ultimaker/Cura #11522: Misaligned font in settings

physicsimulation, 2022-02-21:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Creality Ender-3

### Reproduction steps

Since yesterday (no version change)
1) Opened Cura
Updated software since last few days:
https://www.facebook.com/photo.php?fbid=5122592567752217&set=p.5122592567752217&type=3

### Actual results

All numeric values in settings are up offset.
[see picture](https://scontent.fymy1-2.fna.fbcdn.net/v/t39.30808-6/274576633_5122302887781185_7257948078436839966_n.jpg?_nc_cat=111&ccb=1-5&_nc_sid=5cd70e&_nc_ohc=L7GGWUHCDKQAX-XsGiS&_nc_ht=scontent.fymy1-2.fna&oh=00_AT85aNFqjR8kKZ4umQRdM6WQczN6vM6lK9oaPld8OYCbKw&oe=6218437A)

### Expected results

I am able to modify values by clicking in the value box, below the value.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Updated software since last few days:
https://www.facebook.com/photo.php?fbid=5122592567752217&set=p.5122592567752217&type=3

fvrmr, 2022-02-22:
Hi @physicsimulation thank you for your report.
I haven't seen this before, maybe this is related to your monitor settings, like dpi? 
Did it happen after the update of Cura or your computer?

physicsimulation, 2022-02-22:
I didn't update cura.
I didn't update software manually.
I just reboot windows and some Microsoft software update like Edge. (see
the image list)

physicsimulation, 2022-02-22:
Update ...
I didn't ask to update Office, I just reboot Windows 10.
I found that the misbehavior start around the 8th of february.
I had installed and uninstalled Libre Office during that time.
Uninstalling 4.12 and installing 4.13, no change.
I suspect that Office got "fu** up" (excuse my french).
Then I just "Modify" Office Install, choose "Fast repair" (offline) and Voilà.
Everything is fixed
[image]

fvrmr, 2022-02-24:
Thank you for getting back with me. Good to know that everything is working now.
I will close this issue, but feel free to reach out if you run into any issues again.

closed 2022-02-24 by fvrmr