Ultimaker/Cura #11527: Crash on Clear Bed Plate (Ctrl+D) while writing current code to disk

Balakay612, 2022-02-22:
### Application Version

4.13.1

### Platform

Windows 11

### Printer

Biqu B1

### Reproduction steps

I sliced a model and started saving it to disk; while it's saving, I attempted to clear the build plate (Ctrl+D).

### Actual results

Cura crashes and requires a restart.

### Expected results

Either a dialog saying "Slow down, cowboy!" or something indicating that Cura will clear the build plate AFTER the current write operation is complete: really almost anything other than a crash!

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Traceback:
Traceback (most recent call last):
```
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 1100, in event
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 482, in event
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\Event.py", line 218, in call
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\Signal.py", line 332, in __performEmit
  File "C:\Program Files\Ultimaker Cura 4.13.1\plugins\BigTree3DPlugin\Bigtree3DStore.py", line 314, in _onWriteJobFinished
    self.do_snap(job.getFileName())
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\Utils\Threading.py", line 31, in _call_on_qt_thread_wrapper
  File "C:\Program Files\Ultimaker Cura 4.13.1\plugins\BigTree3DPlugin\Bigtree3DStore.py", line 328, in do_snap
    outdata = outdata + self.overseek()
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\Utils\Threading.py", line 31, in _call_on_qt_thread_wrapper
  File "C:\Program Files\Ultimaker Cura 4.13.1\plugins\BigTree3DPlugin\Bigtree3DStore.py", line 274, in overseek
    outdatar = outdatar + self.overread(QSize(int(tems[0]),int(tems[1])))
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\Utils\Threading.py", line 31, in _call_on_qt_thread_wrapper
  File "C:\Program Files\Ultimaker Cura 4.13.1\plugins\BigTree3DPlugin\Bigtree3DStore.py", line 100, in overread
    img = Snapshot.snapshot(width = msize.width(), height = msize.height()).scaled(msize.width(),msize.height(),Qt.IgnoreAspectRatio, Qt.SmoothTransformation)
AttributeError: 'NoneType' object has no attribute 'scaled'
```

Log:
```
Thread 0x00003e7c (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.13.1\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00005628 (most recent call first):
  File "threading.py", line 306 in wait
  File "zeroconf\__init__.py", line 2534 in wait
  File "zeroconf\__init__.py", line 1715 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00004624 (most recent call first):
  File "threading.py", line 306 in wait
  File "threading.py", line 558 in wait
  File "C:\Program Files\Ultimaker Cura 4.13.1\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x000005a0 (most recent call first):
  File "zeroconf\__init__.py", line 1369 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x000054e0 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.13.1\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00005738 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.13.1\plugins\BigTreeRemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x000054e8 (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00004df4 (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x000045f4 (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x0000078c (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00001d28 (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00002c60 (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Current thread 0x00004070 (most recent call first):
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 398 in _logInfoWidget
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 184 in _createDialog
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 88 in __init__
  File "C:\build\Cura\4.13-exe\build\inst\bin\cura_app.py", line 162 in exceptHook
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 396 in exec_
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 877 in run
  File "C:\build\Cura\4.13-exe\build\inst\bin\cura_app.py", line 236 in <module>
  File "Console.py", line 36 in run
  File "__startup__.py", line 74 in run
```

fvrmr, 2022-02-24:
Hi @Balakay612 thank you for your report.
I am not able to reproduce your issue. But we have a ticket on our backlog that is related to no warning when closing Cura when you are saving a project file. https://github.com/Ultimaker/Cura/issues/11279
So I will link you issue to our ticket.

Ghostkeeper, 2022-03-12:
The crash happens because of this line in your traceback:

```
File "C:\Program Files\Ultimaker Cura 4.13.1\plugins\BigTree3DPlugin\Bigtree3DStore.py", line 100, in overread
    img = Snapshot.snapshot(width = msize.width(), height = msize.height()).scaled(msize.width(),msize.height(),Qt.IgnoreAspectRatio, Qt.SmoothTransformation)
```

This is the `BigTree3DPlugin` which does something after saving the g-code file. It seems to attempt to create a screenshot of the model that was sliced. To do that it wants to know how big the model is so that it can crop the screenshot properly. Since the build plate is empty by the time it wants to make that screenshot, it's not able to get a size. It probably ends up with a size of 0 and crashes because of that.

It's not in Cura's code here. I'd suggest to report this to the maintainers of that plug-in, specifically here which seems to be the same issue: https://github.com/bigtreetech/Bigtree3DPluginSuit/issues/16 . They'll probably want to know the reproduction steps of your crash.

GregValiant, 2025-11-15:
Every once in a while I go through these and close some.  This seems like it was not a Cura problem per se, but rather the plugin.
Can I close this?