Ultimaker/Cura #11533: Editing printer settings

bebr42, 2022-02-23:
### Is your feature request related to a problem?

It should be possible to change - and save - the printer 'machine settings'. 
Currently they can be edited but are not saved when the window is closed. 

### Describe the solution you'd like

Option to save the settings after editing. 

### Describe alternatives you've considered

Adding a new printer profile with the remaining parameters copied over from the original - tedious. 

### Affected users and/or printers

All users and printers.

unknown-loser, 2022-02-23:
They are saved by default when you close the window with the button on the bottom right labeled "close"

bebr42, 2022-02-24:
This does not (always) work. 
See also [this forum thread](https://community.ultimaker.com/topic/30585-cura-not-retaining-custom-start-gcode-in-machine-settings/).

fvrmr, 2022-02-24:
Hi @bebr42 thank you for your feature request.
But this will have the same conclusion as #11502

@unknown-loser is correct that it saves when closing the window.

However the forum thread you linked is more related to this: https://github.com/Ultimaker/Cura/issues/11474. Where the issue is when opening project files will override your machine settings.

bebr42, 2022-02-24:
Sorry - I really had the experience that my 'machine setting' edits did not persist, but can not reproduce the behaviour now. 
Apart from this, it does make sense to save 'machine settings' independent of emf files. 
This is also what the linked thread is about - and not "Reset button to get original Start-/Endgcode back in Printer settings".
So the feature request text shall be changed:
'Save and load printer machine settings' - Cura doen not have it, several users ask for it.

fvrmr, 2022-02-25:
On our backlog it is marked as a new feature with the description prevent overriding machine settings. I will close this issue since it is a duplicate of #11474

closed 2022-02-25 by fvrmr

bebr42, 2022-02-26:
This is really nonsense, sorry for the harsh wording, but more diplomacy does not fit here.
"Editing printer machine settings" and "Reset button to get original Start-/Endgcode back in Printer settings" are abviously two completely different issues.
It seems you can not stand any criticism nor constructive improvements on Cura, so - stay in your pit.
I will stop offering the shovel.

Ghostkeeper, 2022-03-07:
Indeed, it should save those settings when you close the window.

You're not really stating what the problem is you're trying to solve, but this comes close:

> Apart from this, it _does_ make sense to save 'machine settings' independent of .3mf files.

I must read between the lines a bit to deduce what could've happened to you that led to this question. One of the possibilities is that you customised the settings to have different start/end g-codes. Then you loaded in a project file and selected the option to re-use an existing printer, which restores all settings to what they were when the project file was saved, including the start/end g-codes. And you didn't save the start/end g-code before loading that project file, so now they're gone. In this case I'd advise to look at the pop-up that appears when you load a project file. It asks you whether you want to create a new printer, or override an existing one. To prevent losing your existing settings in an existing printer, simply let it create a new printer. Some people find it useful to have a "messy printer" on which they load the project files that they just want to try once. Some people just let it create new printers every time (but then they get a very long list).

When considering features like this, we also have to think of the user experience. More configurability is nice, but if it's just as easy to workaround as to select a new printer when loading a project file, and the alternative is likely to lead to confusion, we'd rather not.

Maybe the more-distilled version of what you're asking is: How can we _merge certain settings from different profiles together?_ If I want to get those machine settings from this profile, but the support settings from another, how can I combine them? There is no real feature to do that yet. It has never been asked as far as I know.

> Cura does not have it, several users ask for it.

If several users asked for it, please point out those existing issues, and add to the discussion there! It's best to keep the discussion about the same issue in one thread, so we can close this one as a duplicate then.

bebr42, 2022-03-07:
Too many assumptions beyond what I actually wrote. 
Simply forget what let my to this proposal, just take it as it is written:

**It should be possible to change - and save - the printer 'machine settings'.**

The other bug "Reset button .. in Printer settings" may be solved by the same feature, but the title is simply misleading.

Ghostkeeper, 2022-04-02:
Fine, just this then.

> It should be possible to change - and save - the printer 'machine settings'.

As far as I can tell, it is possible to change and save the machine settings. You can change them with the checkboxes, dropdowns and text fields in that dialogue, and when you do, it saves that instantly.

bebr42, 2022-04-02:
Please consider reading the forum thread linked above:
"Unfortunately, the options seem to be either to overwrite my current machine settings or else create new machine settings using the settings from the project. It looks like I'm going to have to paste my custom gcode into every old project I load."
So to make it 100% clear - the request is changed to:
"It should be possible to change, save **and load** the printer 'machine settings'."