Ultimaker/Cura #11534: Profils

JLNapo, 2022-02-23:
### Application Version

4.12.1

### Platform

W10 Pro

### Printer

Leapfrog Creatr 2 Heads

### Reproduction steps

The profile does not retain "supports" when switching from one profile to another.

### Actual results

idem

### Expected results

idem

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no file

fvrmr, 2022-02-24:
Hi @JLNapo thank you for your report.
Could you explain a bit more what your issue is?
Do you see _not supported_ at the profiles? Or is there no support generated on your model when using a profile?

JLNapo, 2022-02-24:
Hello,
Sorry for my too simple and poorly explained request.
When I create a profile, I check the box "supports" I put the parameters, and it's supposed to save the profile like this.
If I choose another profile with no supports, when I return to the profile with supports, there are none. Profiles should be used to make choices for repeated use without having to re-parameterize, right?
I don't know if I am using the software correctly.
Thank you for all your efforts.
Cordially,
Jean-Luc

fvrmr, 2022-02-24:
Hi @JLNapo thank you for your reply.
Here you can find more information about switching profiles: https://support.ultimaker.com/hc/en-us/articles/360012381720-Switching-print-profiles-with-unsaved-changes

But what could help you is making your own custom profile (based on a default profile) where the support is enabled. Read here more: https://support.ultimaker.com/hc/en-us/articles/360012380880-The-print-profile-quick-menu-in-Ultimaker-Cura

Let me know if that helps you.

fvrmr, 2022-02-25:
Thank you for your reply. I will close this issue then. Feel free to reach out when needed.

closed 2022-02-25 by fvrmr