Ultimaker/Cura #11537: Cura 4.13.1 Top surface when monotonic enabled.

frankygoop, 2022-02-23:
### Application Version

4.13.1

### Platform

Windows 11

### Printer

Tevo Tarantula

### Reproduction steps

With monotonic enabled, not sure if can be reproduced. only happened on layer 148. The last layer is 150. 

Attached Gcode. go to layer 148.

I also recorded a video https://we.tl/t-7EK7KwkhVe

### Actual results

I believe previous versions did it ok. I just realized after updating to 4.13.1, when monotonic top is enabled, the head move back and forth jumping from one end to another and not printing. 
It is difficult to explain. I recorded a video. The first portion is with the monotonic enabled. the second half is disabled and printing fine. 
I noticed only happened on layer 148. the last layer is 150.

### Expected results

Print continuously along the wall path as normal, not jumping back and forth.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[TT_Africa v7.zip](https://github.com/Ultimaker/Cura/files/8128035/TT_Africa.v7.zip)

fvrmr, 2022-02-24:
Hi @frankygoop thank you for your report.
I think this is a duplicate of https://github.com/Ultimaker/Cura/issues/11401 
Are you using zigzag as top/bottom pattern?

frankygoop, 2022-02-24:
yes, ZigZag.

fvrmr, 2022-02-25:
Then I will close this as a duplicate.
You can change the pattern in Lines and it shouldn't happen.

frankygoop, 2022-02-25:
Dear Fvrmr, Sorry for my confusion with the infill pattern. 
I confirm the mentioned issue is happening with Top/Bottom Pattern in Lines.

Ghostkeeper, 2022-03-12:
Perhaps you could share a project file of that happening, then? You can save a project file with File -> Save Project. It allows us to reproduce the model and the settings you're using.

frankygoop, 2022-03-14:
I attached the project in the first post, go to layer 148. thanks

fvrmr, 2022-03-16:
The attached file is a gcode, to save a project file: file - save project and share it here as a .zip.

Vandresc, 2022-10-14:
Fixed in 5.2.

closed 2022-10-14 by Vandresc