Ultimaker/Cura #11538: Loading Issue?

RealWulaMar, 2022-02-24:
### Application Version

4.13.1

### Platform

Mac OS

### Printer

Creality 3D Ender 6

### Reproduction steps

Downloaded Cura, opened Cura, signed in, updated and installed plugins.

### Actual results

An error pops up telling me to "backup and restore configurations".

### Expected results

Cura should close, then restart, but it does not. Instead it erases and I have to start over everytime.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fvrmr, 2022-02-24:
Hi @5t1ckyF4pD4p thank you for your report.
Does this error shows up everytime you want to start Cura?

5t1ckyF4pD4p, 2022-02-28:
Yes. When I open Cura I sign in and sync up all my plugins. Then, I have to
close out and open up Cura again in order to install the plugins. But,
everytime I get an error when trying to reopen Cura that says Backup &
Restore Configurations (pic below). So, I do that. Then, I try opening Cura
again and it's all been erased. I have to start over again. Ive tried
trashing it and re-downloading the app multiple times, but still no love.

fieldOfView, 2022-02-28:
It looks like there's a problem between one of the plugins you are installing via the sync and the version of Cura you are using. This causes Cura to crash in a way that it thinks the best recourse is to create a zipped backup of your configuration and start clean.

In the "Cura can't start" dialog, press the "Show configuration folder". Then go up one folder and there should be a number of .zip files by now (one for every time you reset Cura via that dialog). Post a link to one of them, and we can have a look at what is going wrong.

fieldOfView, 2022-02-28:
> Then go up one folder and there should be a number of .zip files

I'm not 100% sure if you have to go up one folder. If there are zip files in the configuration folder itself, post a link to one of those.

5t1ckyF4pD4p, 2022-03-06:
Sorry, I thought I posted it a while ago. Here's the link...

Ghostkeeper, 2022-04-02:
Looking at the log, I don't think it's an issue with any plug-in. The PowerManagement plug-in seems to not work for you (it's giving an error when it loads) but it's not the cause of the crash. The cause of the crash is that it's trying to get your account log-in token from the MacOS keyring. MacOS forbids it. This is the relevant part of the log:

```
2022-02-28 12:07:51,449 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: An uncaught error has occurred!
2022-02-28 12:07:51,454 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [71]: Traceback (most recent call last):
2022-02-28 12:07:51,461 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [71]:   File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/site-packages/keyring/backends/macOS/__init__.py", line 49, in get_password
2022-02-28 12:07:51,469 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [71]:   File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/site-packages/keyring/backends/macOS/api.py", line 101, in find_generic_password
2022-02-28 12:07:51,476 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [71]:   File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/site-packages/keyring/backends/macOS/api.py", line 43, in raise_for_status
2022-02-28 12:07:51,481 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [71]: keyring.backends.macOS.api.SecAuthFailure: (-25293, 'Security Auth Failure: make sure python is signed with codesign util')
2022-02-28 12:07:51,486 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [71]: 
2022-02-28 12:07:51,492 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [71]: During handling of the above exception, another exception occurred:
2022-02-28 12:07:51,497 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [71]: Traceback (most recent call last):
2022-02-28 12:07:51,502 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [71]:   File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/site-packages/cx_Freeze/initscripts/__startup__.py", line 74, in run
2022-02-28 12:07:51,508 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [71]:   File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/site-packages/cx_Freeze/initscripts/Console.py", line 36, in run
2022-02-28 12:07:51,517 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [71]:   File "/Users/ultimaker/build/Cura/4.13/build/inst/bin/cura_app.py", line 236, in <module>
2022-02-28 12:07:51,525 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [71]:   File "/Users/ultimaker/build/Cura/4.13/build/inst/lib/python3.8/site-packages/cura/CuraApplication.py", line 844, in run
2022-02-28 12:07:51,530 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [71]:   File "/Users/ultimaker/build/Cura/4.13/build/inst/lib/python3.8/site-packages/cura/API/__init__.py", line 54, in initialize
2022-02-28 12:07:51,536 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [71]:   File "/Users/ultimaker/build/Cura/4.13/build/inst/lib/python3.8/site-packages/cura/API/Account.py", line 118, in initialize
2022-02-28 12:07:51,541 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [71]:   File "/Users/ultimaker/build/Cura/4.13/build/inst/lib/python3.8/site-packages/cura/OAuth2/AuthorizationService.py", line 282, in loadAuthDataFromPreferences
2022-02-28 12:07:51,546 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [71]:   File "/Users/ultimaker/build/Cura/4.13/build/inst/lib/python3.8/site-packages/cura/OAuth2/AuthorizationService.py", line 98, in getUserProfile
2022-02-28 12:07:51,551 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [71]:   File "/Users/ultimaker/build/Cura/4.13/build/inst/lib/python3.8/site-packages/cura/OAuth2/AuthorizationService.py", line 107, in _parseJWT
2022-02-28 12:07:51,557 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [71]:   File "/Users/ultimaker/build/Cura/4.13/build/inst/lib/python3.8/site-packages/cura/OAuth2/KeyringAttribute.py", line 42, in __get__
2022-02-28 12:07:51,563 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [71]:   File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/site-packages/keyring/core.py", line 55, in get_password
2022-02-28 12:07:51,572 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [71]:   File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/site-packages/keyring/backends/macOS/__init__.py", line 55, in get_password
2022-02-28 12:07:51,578 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [71]: keyring.errors.KeyringError: Can't get password from keychain: (-25293, 'Security Auth Failure: make sure python is signed with codesign util')
```

This might give you an idea of a workaround. Not logging into the Ultimaker account would help, or perhaps you need to give some sort of permission somewhere to allow Cura to get the log-in token from your keychain.

This issue is being discussed here: https://github.com/Ultimaker/Cura/issues/9765

5t1ckyF4pD4p, 2022-04-02:
Ok, thank you so much. I did end up deleting a few of the unnecessary
plugins and it's been working fine, so far. But, when I sign in it keeps
wantin to add a couple even tho I've canceled many times. Other than that,
it's been working, tho. Thank you again for your help.