Ultimaker/Cura #11541: Adaptive Layers Issue With Model Off Build Plate

jorlain83, 2022-02-24:
### Application Version

4.13.1

### Platform

Windows 11

### Printer

Creality CR6 SE

### Reproduction steps

1. Raise model from build plate
2. Enable "Use Adaptive Layers"
3. Slice

### Actual results

Application indicates a build time and material usage of 0 and does not slice the model.

### Expected results

Model is sliced and has an estimated build time and material usage.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[AdaptiveLayerIssue.zip](https://github.com/Ultimaker/Cura/files/8133662/AdaptiveLayerIssue.zip)
[image]
[image]
[image]

fvrmr, 2022-02-28:
Hi @jorlain83 thank you for your report.
I will bring it up with the team. Keep you posted.

Ghostkeeper, 2022-03-12:
What was the result of that discussion?

Seems like the same as this issue: https://github.com/Ultimaker/Cura/issues/3183

fvrmr, 2022-03-21:
I forgot to update the status, but we decided to defer this, it has not a high priority to fix this. 
I will close this issue since it is indeed a duplicate of #3183

closed 2022-03-21 by fvrmr