Ultimaker/Cura #11542: Problem with support

MikleD52, 2022-02-24:
### Application Version

4.13.1

### Platform

Win 10

### Printer

Many... in this example sapphire (custom profile)

### Reproduction steps

support on overhanging part
If 50 degrees of support, support does not creates.
If 45 degrees, support exist.

### Actual results

support does not exist when you put 50 degrees on overhanging part

### Expected results

I whant support on 50 degees on overhanging part ;
[image]
[image]
)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

Asterchades, 2022-02-25:
I suspect this to be a combination of two factors.

Firstly, the support angle is the amount from vertical the surface is. 0° is perfectly vertical, 90° is perfectly horizontal. Just at a glance those appear to be 45° overhangs, which would mean 50° is simply too steep an angle to take them into consideration.

The absence of any and all supports on the clearly overhanging corners when set to 50° is most likely due to the Minimum Support Area being too high. The lowest portion of those corners is going to be extremely narrow, resulting in the layer they make up having a very small area. Some profiles - Creality most notably, but quite probably others - impose a minimum area which is simply too great for them to be taken into consideration. The Ultimaker profiles set this to 0mm², which will handle situations like this more appropriately.

Ghostkeeper, 2022-03-12:
Perhaps the Use Towers setting is disabled?

Could you share a project file by which we could reproduce this issue?