Ultimaker/Cura #11543: raft loses smooth top layers

Duf571, 2022-02-25:
### Application Version

14.3.1

### Platform

windows 10

### Printer

flsun qqs

### Reproduction steps

using the fl sun qq to print and cura seems to lose the top smooth layers of the raft when printing . no issue with the ender 3. 

### Actual results

print doesnt stick to raft and is ruinned

### Expected results

i expect it to work properly 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

if i reload cura it works for a few prints then it disappears again

fvrmr, 2022-02-25:
Hi @Duf571 thank you for your report.
Could you share both project files (file - save project) as a zip here?

Duf571, 2022-02-25:
Nothing I do will make this layer stick .  this pla produced a beautiful mask not 2 days ago on this flsun.   Now I get this.

fvrmr, 2022-02-25:
Hi @Duf571 could you post that on Github, I don't see your attachments since you've replied via e-mail.

Duf571, 2022-02-25:
Give it a shot. Not sure

Duf571, 2022-02-28:
Ive tried to attach files and its not working.
Ive tried all settings I can think of ... really ired of your software dropping settings on me. It worked fine for weeks. Now its garbage. Is this your ploy to make be my the full version ?
It wont adhere to the bed. It rips out its own infill. Truly junk

Ghostkeeper, 2022-03-12:
Github removes e-mail attachments. To send attachments, you'd have to go back to [your issue on the Github website](https://github.com/Ultimaker/Cura/issues/11543) to post them there.

There is no "full version" (or non-full version) of Cura. Just the one you're using.

Have to keep in mind that we don't have your printer. We can't see if the print sticks to the raft or not, because on other printers this will be totally different. We can only see if the g-code output is correct. So we can only help out if it can be pointed out what's wrong in the g-code output.

Duf571, 2022-03-13:
I will try again once I'm home

closed 2022-04-04 by no-response[bot]