Ultimaker/Cura #11548: Z Seam Alignment(Sharpest Corner) must not consider fuzzy skin

wolfgangmauer, 2022-02-25:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Any

### Reproduction steps

Set Z-seam alignment to sharpest corner(fuzzy skin off)  and slice
Set fuzzy skin on, now the startpoint from walls are changed

### Actual results

Z-seam alignment(sharpest corner) changes when activating fuzzy skin

### Expected results

Z-seam alignment(sharpest corner) remains the same with or without fuzzy skin

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

For all printers, for all models

wolfgangmauer, 2022-03-03:
Hi @fvrmr,
but that means Z Seam Alignment dont work, alignment will be always random on Fuzzy Skin ?!

fvrmr, 2022-03-04:
I have discussed it with the team and this is indeed expected behaviour, the fuzzy skin places the seam in the sharpest corner of that layer and since the fuzzy skin creates random jitters it will align random. 
We also decided that this is something we do not want to change. So I will close this issue, but let me know if you have questions.

closed 2022-03-04 by fvrmr

wolfgangmauer, 2022-03-04:
No problem, with prusaslicer i can use the seam painting even with fuzzy skin