Ultimaker/Cura #11550: Support first layer speed set to skirt speed not initial layer speed.

SniperBlaize, 2022-02-26:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Ender 3 Pro with SKR Mini E3 V2

### Reproduction steps

1. Turn on skirt and supports
2. Set support print speeds to something different than the skirt print speed and lower than initial layer speed.
3. Slice model
4. Go to preview and select speed. 

### Actual results

For the first layer you will see the support areas will have the same print speed as the skirt. Once it reaches the second layer the support speed goes to what is supposed to be. 

### Expected results

All layers of supports should have the print speed set for supports and not for the skirt. Or if the initial layer speed is less that the support speed then it should follow that speed for better adhesion. 

I was getting bed adhesion issues just for the small support areas and needed to turn the speed down to get them stick. 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]
[CE3PRO_Tap Box Bottom.zip](https://github.com/Ultimaker/Cura/files/8145726/CE3PRO_Tap.Box.Bottom.zip)

fvrmr, 2022-03-03:
Hi @SniperBlaize thank you for your report.
This is happening because you have the Support Brim enabled. This setting will use the Skirt/Brim Speed. 
Disabling the Support Brim the first support will be printed the Support Speed.

Ghostkeeper, 2022-03-14:
There is no way as yet to make the skirt print at a different speed than the brim. I don't think there is a good reason to make that configurable either. Both have to stick to the build plate.

closed 2022-03-17 by no-response[bot]