Ultimaker/Cura #11551: Please add support for CR6-MAX

Rsm61, 2022-02-26:
### Is your feature request related to a problem?

For whatever reason, my PC won't run Creality Slicer, and (for whatever reason) Cura doesn't have the CR6-MAX Profile yet. Can I 
use the CR6-SE profile, and just change the build volume? Not sure what to do.

### Describe the solution you'd like

Add a profile for CR6-MAX

### Describe alternatives you've considered

Using Creality Slicer. But it won't run on my PC.

### Affected users and/or printers

CR6-MAX

fvrmr, 2022-02-28:
Hi @Rsm61 thank you for your report.
We don't maintain these profiles, since the community will contribute these profiles to us. 
However you can indeed change the build volume of the CR6-SE profile. Or use the option custom fff printer.
I will close this since we won't add this profile.

closed 2022-02-28 by fvrmr

fieldOfView, 2022-02-28:
> I will close this since we won't add this profile.

Just to clarify: Ultimaker will not _create_ a profile for this printer, but if someone contributes a profile, Ultimaker will add it to Cura just like they did with all the other 3rd party profiles.