Ultimaker/Cura #11552: Cura 4.12.1 slice preview shows 60% or more walls gone with 0.5 coasting distance with 0.1 nozzle.

Silithas, 2022-02-26:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Ender 3 pro custom marlin firmware

### Reproduction steps

Nozzle 0.1mm
Layer height 0.04
wall thickness 0.102
coasting volume 0.5
minimum volume before coasting 0.5

### Actual results

When i try to slice any model with 0.5 coasting volume, half that layer is cut off no matter how big apparently. Print turns out fine, but preview tells otherwise with 80% of the outer and inner wall being cut away.

[https://i.imgur.com/wkbxz7n.jpeg](url)

### Expected results

Expect the slicer to show the proper coasting distance.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[creality_base_extruder_0_%232_2022_don%27t_fuck_0.1_nozzle_in_progress.inst.zip](https://github.com/Ultimaker/Cura/files/8145862/creality_base_extruder_0_.232_2022_don.27t_fuck_0.1_nozzle_in_progress.inst.zip)

Included file. models sliced as small as 0.102 walls  would allow 2 walls on any print.

fvrmr, 2022-03-03:
Hi @Silithas thank you for your report.
Could you share your project file (file - save project) and share it here in a .zip.

Silithas, 2022-03-03:
Here's the project and model with the same problem.

https://www.dropbox.com/s/oyihfxp361byylv/pika.zip?dl=0 Had to use dropbox as git just errors with "is not included in the list" when making a .zip

Ghostkeeper, 2022-03-14:
So in this project file we're looking at a coasting volume of 0.5mm³. It has a layer height of 0.08mm and a line width of 0.102mm. With a line of 0.08mm high and 0.102mm wide, the 0.5mm³ volume will be printed over the course of `0.5mm³ / 0.08mm / 0.102mm = 61.3mm` of line length. So the coasting should happen in just over 6cm. This is more than the ~4cm circumference of Pikachu/The Rock. The Minimum Volume Before Coasting linearly reduces the coasting volume so that it extrudes at least something, but otherwise the print would almost entirely disappear.

The issue is simply that the tiny layer height and line width makes the Coasting Volume have a relatively big effect. It takes a while to extrude 0.5mm³ with such a small nozzle. But the coasting volume seems to be correctly calculated. I suggest you reduce the volume.