Ultimaker/Cura #11557: muptiple models do not reload correctly

1technick, 2022-02-26:
### Application Version

4.13.1

### Platform

windows 10

### Printer

creality

### Reproduction steps

open multiple models
go to preview mode do calculate tracing
replace model file with new one
cura will ask to reload
click on reload and model is updated correctly
replace multiple model files
cura will ask to reload
click on reload and not all changed models are updated

### Actual results

it updates only some models, not all changed

### Expected results

should update models for all changed files

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

this bug is not new, I have seen it happen in previous versions.

fvrmr, 2022-03-03:
Hi @1technick thank you for your report.
Could you show with screenshots your behaviour? Did you multiply one model in Cura? Or do you have different models you want to replace?

1technick, 2022-03-08:
hi, it was different models one of each. You want see much on screen shot unless you know how it should look like so you have to make something sufficiently different shape to see that it fail.

fvrmr, 2022-03-16:
A video/gif would be nice, because I don't understand what issue you are having. If the file is to big you can share a link via WeTransfer or Google Drive.

Ghostkeeper, 2022-03-20:
These are the reproduction steps I'm trying, which I think shows the problem:

# Create a cube and store it in `moving_mesh.stl`.
# Copy that file to `moving_mesh2.stl`.
# Load both of these into Cura.
# Create a sphere and store it in `moving_mesh.stl` Cura shows a pop-up to ask if `moving_mesh.stl` should be reloaded at this point.
# Copy that file again to `moving_mesh2.stl`. Cura changes the pop-up to ask if `moving_mesh2.stl` should be reloaded.
# Click the button to reload `moving_mesh2.stl`. Only one of the meshes reloads.

Pressing F5 instead in the last step will reload all meshes.

I think the issue is simply with the pop-up, that only one file is shown to need to be reloaded at a time. We could make it pop-up two messages in this case, maybe.

1technick, 2022-03-22:
Looks like you see the problem now. You can either a) show all notification at the same time and allow individual reloads or b) show the last one and always reload all. I am fine with either and you will probably choose b), because of how notifications are implemented and I can not imagine a situation when I would need to keep the old models loaded (it would likely create other complications). In fact I was expecting that to be the behavior and not doing so a bug. If clicking on any reload notification have the same effect as F5 would it would solve it then.

MariMakes, 2024-11-19:
👋 Quick update on our side.
This is resolved in the 5.9 release 🎉 , you can download the Cura version with the fix here: [https://github.com/Ultimaker/Cura/releases/tag/5.9.0](https://github.com/Ultimaker/Cura/releases/tag/5.9.0)

I'll close this issue since it's resolved.
Thanks again, and please let us know if you run into any other issues 💪

closed 2024-11-19 by MariMakes