Ultimaker/Cura #11559: Support Placement Touching Build Plate overrides manually placed support

SQLCraftsman, 2022-02-27:
### Application Version

4.13.1

### Platform

WIndows 10

### Printer

Creality Ender-3

### Reproduction steps

1. Add base model.  Model is figurine with circular base plate, but itcould be any model with z opening.
2. Support Enabled
3. Tree Support Selected
4. Support Placement: Touching Build Plate
5. Add support object.  Can be from plugin or second STL file.  Behavior is identical.
6. Set support object specific property to Print as support
7. Adjust size and placement of support object
8. Support MUST be across part of the base object.
9. Slice

### Actual results

Portion of Support model above any part of the base model is not rendered.  

### Expected results

Manually placed support objects should not be overridden by "Support Placement: Touching Build Plate". 
All manually placed support objects should render as placed
[Cure Support Render bug.zip](https://github.com/Ultimaker/Cura/files/8147914/Cure.Support.Render.bug.zip)
.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[Cure Support Render bug.zip](https://github.com/Ultimaker/Cura/files/8147916/Cure.Support.Render.bug.zip)

fvrmr, 2022-03-04:
Hi @SQLCraftsman thank you for your report.
This is expected behaviour, since you selected touching build plate. This is happening because of your model, since you have that base part in your model. 
Changing it to Everywhere, will print your custom support. Let me know if that make sense to you.

Ghostkeeper, 2022-03-14:
I think the issue here is that this user doesn't expect that to happen. Not so much a bug report then (working as intended) but more of an improvement request (not working as expected). In a way, it makes sense that user-placed support is always printed, even if that causes support to lean on the model. Should we consider improving the UX on this, or perhaps to change this behaviour?

GregValiant, 2024-11-16:
The latest versions of Cura included some reworking of the support structures.  Is this bug report still valid(?) or can it be closed?

SQLCraftsman, 2024-11-20:
The behavior still exists in Cura 5.8.0.

I use a miniature figurine and attempt to place  cube set to "print as support" on the base "coin" for the figurine's legs.   Both tree and normal block with "touching buildplate" enabled.

Bug is still valid.

--Geoff.

GregValiant, 2024-11-20:
I'll leave this open.