Ultimaker/Cura #11565: Successive Scaling of Support Profiles for Greater Efficiency

Obliman, 2022-02-27:
### Is your feature request related to a problem?

I've noticed that when slicing parts the support material is always a simple straight vertical shape that directly maps the area to be supported to whatever is beneath it (when printing in normal mode in Cura 4.13.0).

The following example shows a spiral staircase that has an excessive amount of support material compared to what could be used.
[image]
 
I see that there is an experimental conical support mode, but it does not address this situation well since it just "tucks-in" the bottom of the support:
[image]
 

### Describe the solution you'd like

Instead, if the area to be supported (and the support area on each layer) is representable as a closed polygon, then a method like polygon buffering (AKA inward/outward polygon offsetting) can be used to generate successive scaled polygons that lie within the original outline, like so:
[image]

This would reduce the amount of support material used and print time, while still allowing for complex geometry to be supported.

Additionally, it may be desirable to subdivide a support area into smaller sections before scaling, creating a structure of "arches":
[image]
[image]

A minimum area can be defined to prevent supports from becoming too flimsy/disappearing as they are scaled. Scaling rate per layer, minimum width (to prevent very thin supports), and number of unscaled layers before scaling begins (to improve quality at the top support layers) could also be useful parameters.

For the base, it may make sense to generate one of the following:

- No additional material
- A brim/miniature raft for better adhesion to the print bed
- An approximate inverse of the top where it scales back to the original size, like so:
[image]

I generated the above images by mocking-up the idea in Python using the [shapely](https://pypi.org/project/Shapely/) library, which handles the polygon scaling and area calculations.

### Describe alternatives you've considered

There are a lot of possible alternative methods for generating supports. I've made use of the density gradient settings for support material that step the density up as you approach the material to be supported, which this method could further improve upon.

### Affected users and/or printers

Potentially everyone could benefit from a more efficient support material method, particularly where there are large vertical spans of support that don't really need the full volume filled.

### Additional information & file uploads

I'm making a few assumptions about how Cura handles the designation of support areas, but I have heard that it creates zones to further process with infill patterns, walls, etc. It sounds like a relatively easy thing to implement if those zones can be scaled, but please let me know if this is unrealistic.

Obliman, 2022-02-28:
After thinking about this a bit more it seems like a much more complex task when you have a sloped support surface, or many adjacent ones like the stairs. A combination of not scaling the supports near supported features while scaling and combining elsewhere would be required for this to actually improve print time/material usage...

Example of what should not be done, since it probably increases print time over the original support style by adding many support perimeters:
[image]

fvrmr, 2022-03-11:
Hi @Obliman thank you for your request. 
I will bring it up with the team. Keep you posted.

pkuiper-ultimaker, 2022-03-17:
Hi @Obliman,

We reviewed the ticket in both the Cura and the print profile team. We like the idea. We do need to figure out how to deal with tilted surfaces etc. Also splitting the support in to many segments will cause islands, which in turn can cause more retracts and walls which might even lower the productivity. We would need to test that.

Anyway we were planning to improve our support structures (speed and quality) after the next Cura release. We will brainstorm on what to do and take your input along and keep you posted on our discussion.
Kind regards,

         Paul Kuiper

For internal reference: PP-139

Obliman, 2022-03-17:
Thank you for the update. I really enjoy the features and flexibility of Cura, so I'm glad that I can contribute something.

I didn't mention it before, but another concept in a similar line of thinking was to morph the profile of the support area from the exact material shape at the top to something more efficient like an appropriately sized rectangle/circle/smoothed version of the geometry to improve print speed. Again, not a terribly difficult task if you're dealing with just polygons, so I hope such a feature could be feasible.

pkuiper-ultimaker, 2022-10-04:
Hi @Obliman,

Sorry that I did not respond for a while. I think that a large part of your request can be done using conical support, which you will find under experimental:

[image]

The default settings should be OK for most cases. Make sure that you have a wall around the support otherwise you can create loose hanging support structures when the support starts scaling inwards. Another way to prevent this is changing your support infill direction each layer (e.g. [0,90]).

I hope this helps. 
Kind regards,

Paul Kuiper