Ultimaker/Cura #11566: Save As...

Tone71, 2022-02-27:
### Is your feature request related to a problem?

I want to keep the settings from one model and use them on another. Add model, delete old one, fine. But then I can't 'Save As..' the new model. Very annoying, and more than once I've overwritten the original model.

### Describe the solution you'd like

Save As is pretty standard, for well understood reasons

### Describe alternatives you've considered

.

### Affected users and/or printers

All Cura users will benefit.

fieldOfView, 2022-02-27:
Did you try "Save project..."?

skelliam, 2022-03-06:
@fvrmr @fieldOfView "Save project..." (and the associated CTRL-S) saves the current project.  It is true that this option will ALSO ask to "Save as..." even if you just want to save changes to the current project.  It is more common that CTRL-S will save changes immediately to the open project with no prompt, while "Save As..." will save the entire project under a new name.

Summary:

* Save: update current file (hotkey CTRL-S is common) immediately without questions.  If there is no current file, the Save As behavior below is followed.
* Save As: save entire project under a new name

Microsoft Excel file menu for example:

[image]

Ghostkeeper, 2022-03-14:
I think the root of this issue for the OP is the project name (as shown in the bottom left). It takes on the name of the first model you load, and this name then gets used as the default filename when you save. When you load `file1.stl` the project name will become `file1`. If you then load a second object `file2.stl`, you have multiple models on the build plate. The project name remains `file1`. Even if you delete `file1.stl` from the build plate, it remains `file1`. It doesn't change the given project name when a file is deleted that the project might have been named after.

On the other hand, if you first delete `file1.stl` and then add `file2.stl`, it will take on the project name `file2`. When the build plate is made empty, the project name is erased, and then it can take on a new project name. So changing the order of operations here changes the result such that Save As will save it as the new file.

So I think this is a duplicate of https://github.com/Ultimaker/Cura/issues/9189.

I don't think Tone71 had the problem of accidentally using Save instead of Save As and getting his file overwritten that way, since there is no real "save" option in Cura at the moment. It'll always pop up a dialogue. Skelliam's suggestion might be useful as a separate feature request, but should probably be posted in a separate issue to track.

closed 2022-03-18 by no-response[bot]