Ultimaker/Cura #11567: Cannot slice 3mf file

ruredi87, 2022-02-27:
### Application Version

Arachne beta 2

### Platform

windows 11

### Printer

ender 3 v2

### Reproduction steps

I cannot slice this file with my abs profile. When I try with PLA profile ıt is slicing succesfully. Also if I convert my file to stl file from 3mf It can slice succesfully also.

### Actual results

It reports error

### Expected results

Be able to slice file and save gcode.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]

Here is project file. I could not find log file.
https://drive.google.com/file/d/1kpwNY0Diuce34zKROTQ9tBm8HUgpi6-c/view?usp=sharing

This is 3mf file which I try to slice
https://drive.google.com/file/d/1gx3yZzeZWpi2pr46gmKUAFSBlZtMI-hF/view?usp=sharing

GregValiant, 2022-03-01:
I opened the project file and it sliced.  4.13.1 and Ender 3 Pro with Win 10 Home

fvrmr, 2022-03-03:
Hi @ruredi87 thank you for your report.
I had no problems with slicing your project file in Arachne. Was the file you shared the file that didn't slice?

closed 2022-03-18 by no-response[bot]