Ultimaker/Cura #11568: Non-Ultimaker printers print 100% infill at half speed

richfelker, 2022-02-28:
### Application Version

All

### Platform

All

### Printer

Any

### Reproduction steps

Set infill to 100% using any printer whose definition derives from `fdmprinter.def.json` rather than having its own complete fork of the same.

### Actual results

100% infill is printed as 999999 bottom layers rather than as infill, thereby using the top/bottom speed, acceleration, line widths, etc. This is due to the logic in the value for `bottom_layers` which reads

```
999999 if infill_sparse_density == 100 else ...
```

which does not appear in the Ultimaker printer definition files. Instead they set `infill_pattern` to `zigzag` when doing 100% infill.

### Expected results

Infill should be printed as infill the same way it's done on Ultimaker printers.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Credit for finding this goes to Chuck Hellebuyck; see video on the topic: https://www.youtube.com/watch?v=Z0dW5r1EVEM

Since it seemed weird, I trawled through the printer definition json files to find out why it was happening, and the above is what I found.

fvrmr, 2022-03-03:
Hi @richfelker thank you for your report.
I will bring it up with the team. Keep you posted.

64bittuning, 2022-03-11:
@richfelker
@fvrmr 
This is unfortunately not true, and stems from Chucks lack of understanding. The Ultimaker S3 "fast" profile that he called out in his video, used 50mm/s for all print path movements. These would be Print Speed, Infill Speed, Wall speed (inner/outer), Top Surface Skin Speed, and Top/Bottom Speed. 
Meanwhile, the Ender2 profile set to "Low Quality", uses a Print Speed and Infill Speed of 50mm/s but a wall speed and top/bottom speed of 25mm/s. Even his own profile, not anything built into Cura or even Ultimakers responsibility, uses a print speed of 60mm/s, but still keeps the top/bottom speed of 30mm/s (still half, because the setting in the Cura base config file is "speed_print / 2").

This isn't a bug, its a very unfortunate case of the end users not understanding that Ultimaker is not responsible for updating anything in the "other" profiles, aside from the bare minimum to keep them compatible with Cura. Like changing version settings and such.

richfelker, 2022-03-11:
I thought I was clear in the report that Chuck did not understand how this was happening but that there is an actual root cause where Cura is doing the wrong thing. You can reproduce this without any special profiles, just running `CuraEngine` from the command line with the baseling `fdmprinter.def.json` (where default `speed_topbottom` is 30 while `speed_infill` is 60).

I'm not aware of whatever your policy on "Ultimaker is not responsible for updating anything in the 'other' profiles" is, but the rules in `fdmprinter.def.json` are often modified as part of fixing unwanted behaviors. Regardless of whether this is intentional (I don't believe it is), leaving a known, inexplicably-weird corner case behavior (treating 100% as special and printing it with the wrong extrusion type) in place when it's been shown to yield slow prints (as well as all sorts of other possibly wrong behavior, like violating the expected printing order or printing with the wrong extruder) seems like a really bad look for Cura. This should be fixed.

I've edited the title of this issue to be more clear about the underlying mechanism and not focus on Chuck's manifestation of it.

64bittuning, 2022-03-11:
@fvrmr 
Proposed solution:

Modify fdmprinter.def.json line 2962 
From
`"value": "speed_print / 2",`

to

`"value": "speed_infill if infill_sparse_density == 100 else speed_print * 0.5",`

This will trigger the slicer to apply the infill speed to all top/bottom layers when the infill is set to 100%. infill layers will still be treated and colored as top/bottom layers in the layer view, but the speeds will be corrected and ultimately achieve the same goal without extra steps.

richfelker, 2022-03-11:
That is not a good solution at all. It changes the top/bottom speeds, which the user likely intends to be lower to get a better surface finish (especially if extrusion capability is only borderline at the full speed requested).

The solution is making the same change that was made to the Ultimaker profiles: to stop using bottom layers to fake 100% infill and instead just switch to zigzag pattern for the infill when it's set to 100%. Then infill layers will show and be treated as infill, as they should be, regardless of whether infill is 10%, 25%, 99%, or 100%.

64bittuning, 2022-03-11:
I see. I better understand the concern now, and I agree. Infill should be infill, not top/bottom layers.

removing `0 if infill_sparse_density == 100 else ` from the "top_layers" value field
removing `999999 if infill_sparse_density == 100 else` from the "bottom_layers" field
and changing infill_pattern to `{ "value": "'zigzag' if infill_sparse_density > 89 else 'lines' if infill_sparse_density > 25 else 'grid'" },`
seems like it would solve this issue then. Do you agree?

Ghostkeeper, 2022-03-14:
> I'm not aware of whatever your policy on "Ultimaker is not responsible for updating anything in the 'other' profiles" is

Ultimaker's policy is normally to try to keep changes in the g-code output to a minimum. If changes are necessary we tend to stay on the safe side. New features should be disabled by default then. We don't know what the effect will be on other printers. So the effect is indeed that Ultimaker keeps the hands off as much as they can, until they get a PR or a strong sound that the profiles should be adjusted.

> I'm not sure why Ultimaker chose to revert this for their machines, but the behavior of making 100% infill act as bottom layers has been present since 4.5
It's not new.

Using skin instead of infill at 100% was originally seen as an improvement, because it eliminates a border between the skin and the infill. The border is a weak point, so if you still use infill, your print will be weaker, with some (more) air bubbles in the middle. For some of Ultimaker's printers we recently had some profile optimisation and they determined that the extra printing time is not worth it, and they'd rather print with the infill settings at greater speed, than the skin settings at greater strength.

We could change that also for other printers. The problem is then that we're changing the profiles for hundreds of printers and have no way to test if the changes work there. Settings are finnicky and don't always work the same way on all printers, all materials, all nozzles, etc. and it'd be mean to pull the rug under the people that optimised their profiles for this.

We can still change the default of course, but it'd help if we get some reports on whether it prints well on various printers to get an idea on what's out there on the field. And even then it's a pretty scary decision to make!

That YouTube video sparked some conspiracy theories recently on Reddit of people thinking there was some "secret algorithm" going on that's only intended for Ultimakers. There's not; you guys are exactly right about that this setting formula is the cause.

GregValiant, 2024-11-16:
Things appear to have been changed in later versions of Cura (5.8.0 and up).  Can this be closed or is it still an issue?

closed 2024-12-09 by github-actions[bot]