Ultimaker/Cura #11569: Section analysis

jxer, 2022-02-28:
### Is your feature request related to a problem?

Struggling with Tree support, need to see inside of object on z axis

### Describe the solution you'd like

provide viewing option in , preview that allows cut in half on z  axis, at midplane or make adjustable

### Describe alternatives you've considered

no way to see inside layers. top view does not provide ample viewing of support structures.

### Affected users and/or printers

all users

fvrmr, 2022-03-03:
Hi @jxer thank you for your request.
Could you explain why the layer slider would not work? 
[image]

jxer, 2022-03-04:
because i need it to be sliced in half on the z axis. I am having trouble with tree supports actually supporting things.

see https://github.com/Ultimaker/Cura/issues/11570

although i have used this, a movable z cut would be highly appreciated as well. its called section analysis in fusion360, and its something i use constantly

fvrmr, 2022-03-07:
Thank you for your reply, I will bring it up with the team.
Keep you posted.

Ghostkeeper, 2022-03-20:
Something similar was once implemented as a research project here, although for the solid view instead of layer view. This worked by basically allowing the user to change the near-plane of the render operation, which causes everything closer than a certain distance from the camera to disappear. 3D models are hollow though, including layer view. So for that it also closed up the cut surface, although I don't exactly remember how. This probably wouldn't translate to layer view well, since that 3D structure is generated in the GPU. Something else would need to be implemented, especially considering we do backface culling in layer view.

This could possibly be implemented in a plug-in, as well.

jxer, 2023-06-29:
I would still appreciate this feature.