Ultimaker/Cura #11570: Tree Support, not providing same support as normal support

jxer, 2022-02-28:
### Application Version

4.13.1

### Platform

10

### Printer

ender 3

### Reproduction steps

simply changing support type, changes support coverage.

### Actual results

tree provides far less support then normal, even though over hang does not change.

### Expected results

i would expect same coverage between both, significant difference

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

https://i.imgur.com/gzhKgnh.png tree

https://i.imgur.com/QD1Ihh9.png normal

[image]

fvrmr, 2022-03-07:
Hi @jxer thank you for your report.
Tree support does not mean that it will have the same support as normal support. 
Read more here about the support types: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/support/support_structure.md As you can read there: 

> Tree support often uses way less material than ordinary support. Between 25% and 50% of the material usage is common. This saves a lot of time and reduces the cost of the print.

Ghostkeeper, 2022-03-14:
Although tree support saves time in total, it's still supposed to support everything that has a steeper overhang angle than the Support Overhang Angle setting.

We have made a bugfix for the next release to the way the X/Y and Z distance applies to tree support. If you can provide a project file, we can see if this issue still occurs. Or you could try it in the beta, when that gets released.

You can also try to fiddle with the X/Y/Z distance to see if that helps you get the correct result.

jxer, 2022-03-14:
@Ghostkeeper Thats what i would have thought as well. Is there a way to supply the project privately? the stl is an item sold on marketplaces.

fvrmr, 2022-03-16:
Hi @jxer you can send a email to f[dot]vermeer[at]ultimaker[dot]com