Ultimaker/Cura #11574: Support Interface Transition

Benjamin4456, 2022-03-01:
### Is your feature request related to a problem?

I've been finding that a particular soluble support material I've recently switched to does not adhere well to the sparsely spaced support infill lines.  Presently the only solutions I see are to increase the support density and use gradual infill (not ideal), or to print the entire support from the soluble material rather than just the interface (potentially expensive).

### Describe the solution you'd like

By adding the option to print a support interface transition layer this could greatly improve the adhesion between the interface material and the rest of the support.  I realize this is a somewhat fringe case, but the ability to add a one or two layer transition made from the main material - giving the interface a solid bed to build on and adhere to - should in theory greatly reduce adhesion related failures.

The only downside I see of the support interface transition method is that it would slow solvent access to the soluble material.  This could be controlled by allowing a interface transition density modifier as well.

Support interface transition settings in general would also be nice, including initial layer speed, flow, etc. as well as interface patterns that are non-intersecting and without abrupt direction changes (outside-in concentric for example).

### Describe alternatives you've considered

As mentioned in box one, using gradual supports with a high infill density or printing the entire support (rather than only the interface) from the soluble material are possible alternatives, however neither is a great solution.

### Affected users and/or printers

This would affect any dual extrusion capable machine running a main component material and a breakaway and/or soluble support material.

Ghostkeeper, 2022-03-20:
As I understand the feature request, the issue is precisely with using the support interface though, because the problem is described as being *the layer between the support interface and the rest of support*:

> this could greatly improve the adhesion between the interface material and the rest of the support

The problem is that the support infill and support roof don't stick well enough to each other. So using the support roof extruder is the root cause of the issue then, not a workaround. The best workaround that I can think of is indeed to use gradual infill.

I've never heard of or experienced the adhesion between these to really be problematic though, at least not with PVA. Are you using two different types of plastic for this?

Benjamin4456, 2022-03-20:
> As I understand the feature request, the issue is precisely with using the support interface though, because the problem is described as being _the layer between the support interface and the rest [...]

You're correct that gradual infill can achieve a workable result.  A while after posting this I was able to use high support density with gradual steps to achieve essentially what I was aiming for, however it seems a little clunky.  Realistically though this probably isn't an issue many people encounter, so it being clunky isn't so terrible.

The problem I was having was with the adhesion between ASA and AquaSys 120 support filament, and I still have not gotten it to work reliabily.  This is the first time I've had an issue with support interfaces and other support materials stick mostly fine, so again, it is a fringe case.  I figured the feature would simply provide an easy means to increase the reliability of support interfaces adhering properly.

While it would be nice to see a dense support interface transition layer (between the support structure and the support interface), gradual support works well enough and I don't see this as being a common enough problem to perhaps invest the time in adding the feature.