Ultimaker/Cura #11575: Move Flow % from profile to Material Profile

antv311, 2022-03-01:
### Is your feature request related to a problem?

There is no issue directly more of an annoyance of having to make a ton of profiles for each of my filaments after creating a material profile for each of my filaments.

### Describe the solution you'd like

So currently flow rate % is located only in the material section of the profile. I was wondering if you could also please, add it to the Material Profiles as well. 

Since Flow rate percentages can range from filament types and even sometimes brands it would be nice to be able to set that in the material profile, print settings (see images I know I use the word profile a lot) .  I think it would be a great addition to the this section and would benefit not only those who tune all of their filaments,  but also people that may have a ton of exotic filaments that may not be used everyday. 

Currently the only way to kind of get something similar is to remake all of the material profiles into regular profiles which doesn't seem like this is what the regular profiles are supposed to be used for.  Profiles I thought were more of group of settings for a particular print type,  (ie Miniatures , busts ,  statues, tools , etc ) not for individual materials.  

### Describe alternatives you've considered

I guess I could also just slap a post it on the side of the filament but then I would have to remember to add it on every thing I print which not gonna happen.  I can also remake all of my custom materials into custom profiles. 

### Affected users and/or printers

Well all of those people who tune the filaments will use this defiantly. Anyone that likes to keep digital back of things like settings will. People who have bad memories, or who have no access to post its or permeant markers.  

### Additional information & file uploads

[image]
[image]

I looked around and didn't see this request if I am duplicating an ask I apologize.  Also thank you for all the hard work you guys put into this software , it is really great and intuitive. Thank you.

fieldOfView, 2022-03-01:
See the plugin "Material Settings" in the Marketplace, which lets you add the Flow setting (and many others) to the Print Settings on the Materials pane of the preferences.

GregValiant, 2022-03-01:
Or you can do what those of us with bad memories do and just..........hold on.......damn, I just had it to.

antv311, 2022-03-02:
Thank you both

closed 2022-03-02 by nallath