Ultimaker/Cura #11576: [4.13.1] Strange pathing. Nozzle visits a point without doing anything, before continuing to next area. Combing off.

printingotb, 2022-03-01:
### Application Version

4.13.1

### Platform

win10

### Printer

custom fff

### Reproduction steps

Not sure, see the attached project. Combing is set to off.

### Actual results

I can't tell if this is an issue in arachne since it uses so vastly different pathing so I'll post this just in case.
Nozzle visits a point without doing anything, before continuing to next area. 
4.11 does not show this behavior (if I managed to exactly recreate the settings profile that is).

### Expected results

Skip this extra travel move.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]
[image]
[pathing.3mf.zip](https://github.com/Ultimaker/Cura/files/8160328/pathing.3mf.zip)

smartavionics, 2022-03-01:
I think it's due to monotonic ironing order being enabled. Looks like it's travelling to where it thinks the ironing will start but then the monotonic ordering kicks in and it goes off elsewhere.

printingotb, 2022-03-02:
Yep, doesn't happen if I turn off monotonic ironing.

fvrmr, 2022-03-07:
Hi @printingotb thank you for your report.
I think this is related to the other issue related to Monotonic Ordering you reported a while a go: https://github.com/Ultimaker/Cura/issues/10824 and we have a ticket on our backlog related to the travel moves with monotonic. I will link this ticket as well.