Ultimaker/Cura #11578: infill

cptblue, 2022-03-01:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Elegoo

### Reproduction steps

Unsure but Infill option now does not change when chnaging the percentage in the settings seems to always stay on a default 

### Actual results

Infill doesnt change when chnaging the percentages

### Expected results

Infil to change

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[N2D_potout_no_label.zip](https://github.com/Ultimaker/Cura/files/8163462/N2D_potout_no_label.zip)

GregValiant, 2022-03-01:
"Changing the percentages"
Percentage of what?  Infill Density?
Your printer comes up as an "Unknown Machine" so I can't open the 3mf file as a project.

cptblue, 2022-03-01:
Yeah infill density. Can change percentage from 0-100% but makes no difference in preview or an actual print. It will always look like the default 25% density. 
Elegoo Neptune 2d ended 3 clone really. It went like this when I updated to 14.13.1 as it was fine before.

GregValiant, 2022-03-01:
What printer definition did you use to install your printer in Cura?
If you install an Ender 3 and then play with the Infill Density does the problem still occur?

cptblue, 2022-03-01:
I haven’t tried that but I will. The way I installed is via this git. 
https://github.com/Toylerrr/ELEGOO_Neptune2_Cura

cptblue, 2022-03-01:
ok - sorry using ender 3 profile it works, can set any infill and it changes as per the pecentage selected so must be issue with the elegoo support plugin from Toylerr
https://github.com/Toylerrr/Elegoo-Neptune-Support

fvrmr, 2022-03-03:
Hi @cptblue thank you for your report.
Please report your issue then to the maintainer of the plugin. 
I will close this issue. But feel free to reach out if you have questions.

closed 2022-03-03 by fvrmr