Ultimaker/Cura #11580: Slicing failed with an unexpected error

GameKyuubi, 2022-03-01:
### Application Version

4.13.1

### Platform

Archlinux Nvidia

### Printer

Monoprice MP10

### Reproduction steps

- Load model
- Try to slice

### Actual results

"Slicing failed with an unexpected error."

### Expected results

Slicing should complete successfully.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[CFFFP_chain01.3mf.zip](https://github.com/Ultimaker/Cura/files/8163751/CFFFP_chain01.3mf.zip)

[cura.log](https://github.com/Ultimaker/Cura/files/8163769/cura.log)

Toddi3D, 2022-03-05:
Same Problem here

fvrmr, 2022-03-07:
Hi @GameKyuubi thank you for your report.
I could slice your project file with no issues.
[image]

Are you using the AppImage?

Toddi3D, 2022-03-07:
Its not my fiie

fvrmr, 2022-03-07:
@Toddi3D could you share your file in the other thread you created: https://github.com/Ultimaker/Cura/issues/11598?

GameKyuubi, 2022-03-12:
> Hi @GameKyuubi thank you for your report. I could slice your project file with no issues. 
> Are you using the AppImage?

I installed from AUR, version 4.13.1-1

fvrmr, 2022-03-16:
@GameKyuubi we don't support that version, only the AppImage you can find here: https://github.com/Ultimaker/Cura/releases/tag/4.13.1
Please report your issue to the maintainer of that version. If you run into issue with the AppImage let us know.

I will close this issue.

closed 2022-03-16 by fvrmr